analogdevicesinc/libiio #150: libusb in context

rgetz, 2018-02-21:
to help debug some libusb issues we were having, I added this code:

```
diff --git a/usb.c b/usb.c
index 2a48744..e5ef23e 100644
--- a/usb.c
+++ b/usb.c
@@ -699,6 +699,7 @@ static int usb_populate_context_attrs(struct iio_context *ctx,
                const char *attr;
                uint8_t idx;
        } attrs[3];
+       struct libusb_version const *ver;
 
        libusb_get_device_descriptor(dev, &dev_desc);
 
@@ -740,6 +741,14 @@ static int usb_populate_context_attrs(struct iio_context *ctx,
                }
        }
 
+       ver = libusb_get_version();
+       iio_snprintf(buffer, sizeof(buffer), "%i.%i.%i.%i%s %s",
+                       ver->major, ver->minor, ver->micro,
+                       ver->nano, ver->rc, ver->describe);
+       ret = iio_context_add_attr(ctx, "local,libusb", buffer);
+       if (ret < 0)
+               return ret;
+
        return 0;
 }
```

which seems to be pretty straightforward, I just wanted to understand the preference - "local,libusb", or "usb,libusb", or something else? I expect something else will need to go into the iiod/usbd.c:usb_open_pipe() to add the remote version as well.

the above prints out:

```
analog@imhotep:~/github/libiio$ iio_attr -a -C
Using auto-detected IIO context at URI "usb:3.2.5"
IIO context with 8 attributes:
local,kernel: 4.6.0-08783-g1cc7526
usb,idVendor: 0456
usb,idProduct: b673
usb,release: 2.0
usb,vendor: Analog Devices Inc.
usb,product: PlutoSDR (ADALM-PLUTO)
usb,serial: 104473222a8700071a002b001f4dd56fb8
local,libusb: 1.0.21.11156 http://libusb.info
```

pcercuei, 2018-03-08:
The prefix should be the backend that corresponds to the attribute, so it should be "usb,libusb".

No need to modify iiod, the remote will never use libusb (it uses FunctionFS).

rgetz, 2018-05-09:
So, did you want a pull request for this?

pcercuei, 2018-05-11:
That would be great, yes.

rgetz, 2018-10-25:
This is tracked in #207 now

closed 2018-10-25 by rgetz