analogdevicesinc/libiio #154: C# Binding - IIOBuffer.fill() Missing Functionality

raceybe, 2018-03-07:
The functionality provided by the fill() method uses the Marshal.Copy command:

`Marshal.Copy(array, 0, iio_buffer_start(buf), length);`

However, this is only good for copying from the managed array to the unmanaged OUTPUT buffer.

Another function is required for input buffers (or perhaps a flag to select the appropriate Marshal.Copy overload). For an INPUT buffer, we need to copy from unmanaged buffer to managed array...something like the following:

```
public void fillFromInput(short[] array)
{
long length = (long)iio_buffer_end(buf) - (long)iio_buffer_start(buf);
if (length > array.Length)
length = array.Length;
Marshal.Copy(iio_buffer_start(buf), array, 0, array.Length);
}
```

Notice the new order of the input arguments.

pcercuei, 2018-03-07:
Indeed, we seem to lack a function here. Feel free to send a patch, and we can add it (note: I'd prefer IOBuffer.read() as the function name).

By the way, you can extract data from the IOBuffer using Channel.read(), and it gives you the samples demultiplexed for that channel. Which might not be what you want to achieve, I just thought I'd mention it.

raceybe, 2018-03-07:
I will look into the Channel.read(). Does the same thing apply to Channel.write() as well? That may just render a days work obsolete.

it appears read() and write() operate on a byte level, so the 16 bit word would still need assembly from the stream? I had modified/added my function as a short so the entire 16 bits could be read/write in one shot.

pcercuei, 2018-03-07:
The same thing applies to Channel.write() as well, yes.

Channel.read()/write() use 8-bit arrays, but you're free to reinterpret (cast) them as 16-bit arrays. This will work as long as the hardware expects 16-bit samples.

rgetz, 2020-01-14:
@AlexandraTrifan : can you have a look at this?

rgetz, 2020-02-04:
Pull request merged - so closing this.

closed 2020-02-04 by rgetz