analogdevicesinc/libiio #158: LIBIIO access via network on Windows Subsystem for Linux

jaredd, 2018-03-09:
I've been trying to get my ADALM-Pluto to work over the network context on windows subsystem for linux on Win 10 Pro.  The PlutoSDR is attached through a powered USB hub.  I setup the device according to the analog devices tutorials.  The device operation was verified using the ADI Oscilloscope on windows.  The windows iio_X utilities also work.  

I would like to use Windows Subsystem for Linux for its ease of GNU Radio development.  The libiio library and utilities were built using native (linux) tools on WSL.  The [libiio_git_diff.txt](https://github.com/analogdevicesinc/libiio/files/1798242/libiio_git_diff.txt) patch seems sufficient to get proper operation on WSL.  I'll describe what symptoms indicated a problem and how the patch seems to fix them:

1.  iio_info -n 192.168.2.1 would hang (not exit or error) despite being able to both ping and ssh to the device from WSL.  I'd expect, if there was a problem, that a problem would be indicated and the utility would exit.  In the patch, the lines negating the network_should_retry call seem to result in the right behavior.  It's hard to tell whether this is the right behavior in general, it seems to me if "network_should_retry" returns False then you should return and not retry.  In the repo the behavior is the opposite.

2.  After fix 1, the context could not be created and the utility would exit.  It seems that windows doesn't have good support for MSG_TRUNC through WSL in the network_read_line function.  I simply added an additional __WSL__ define and forced the non linux code to be used instead.  I'm not sure how a good way to auto-detect WSL in CMake, though "Microsoft" is in the uname line below.  Cmake seems like the right place to detect and set defines.

More info.  Development was from a fresh clone of libiio.

$ git remote -v
origin  https://github.com/analogdevicesinc/libiio (fetch)
origin  https://github.com/analogdevicesinc/libiio (push)

$ git describe
v0.7

$ uname -a
Linux WPL-T470 4.4.0-43-Microsoft #1-Microsoft Wed Dec 31 14:42:53 PST 2014 x86_64 x86_64 x86_64 GNU/Linux

pcercuei, 2018-03-10:
Interesting, I think you are the first one to try libiio on WSL.

For your issue 1, please try the latest libiio from master. The code you patched has changed since then.

For issue 2, we can just hardcode the define, I think, because we know the exact configuration where it works. If you want credit, send a PR with a fix :)

lclausen-adi, 2018-05-07:
Fixed with #171.

closed 2018-05-07 by lclausen-adi