analogdevicesinc/libiio #161: scopy opt in - end user usage monitoring

rgetz, 2018-03-27:
In order to see what instruments people are using, and what we should focus on more, we need to have the option for an opt-in, automatic statistics gathering. This could capture things like:
 - time each, and how long each instrument is "run", uptime of the device (how long it's plugged into USB)
 - operating system, screen resolution, memory usage
 - scopy version, hardware version, firmware version
 - country (if possible)
 - Where you click (to see if there are common clicks everyone does in a certain order)
 - min/max max temp of the zynq and ADC
 - basic info about specific instruments, 
   - ie - function generator - which waveform are being used, what voltages, what frequencies
  - spectrum analyzer, what start/stop frequencies 

Crash information would be a good one as well.

Firefox/Mozilla already does this, and calls it telemetry :
-  https://firefox-source-docs.mozilla.org/toolkit/components/telemetry/telemetry/
 - https://wiki.mozilla.org/Firefox/Data_Collection#Requesting_Approval
 - https://support.mozilla.org/en-US/kb/send-usage-data-firefox-mobile-devices
Chrome "Usage statistics and crash reports":
 - https://www.google.com/chrome/browser/privacy/whitepaper.html
There is a discussion on how Microsoft encourages developers to do this at:
 - https://blogs.windows.com/buildingapps/2014/03/20/instrumenting-your-app-for-telemetry-and-analytics/

A great white paper talking about how google implement crowdsourcing statistics from end-user client software, anonymously, with strong privacy guarantees is at:
- https://arxiv.org/abs/1407.6981
 and code is at (where there is a cpp client):
 - https://github.com/google/rappor

I think this would make a great co-op project ???

-Robin

lclausen-adi, 2018-03-27:
Was this meant for the Scopy issue tracker?

rgetz, 2018-05-09:
yes. closing - sorry - I moved this over.

closed 2018-05-09 by rgetz