analogdevicesinc/libiio #164: Running dummy example gets zero device count

mortenfyhn, 2018-04-12:
Hi all. I'm a bit confused about how to run the [dummy example](https://github.com/analogdevicesinc/libiio/blob/master/examples/dummy-iiostream.c). It compiles fine, but crashes because `iio_context_get_devices_count(ctx)` returns zero. There instructions for how to set the dummy device up are written as
```
 *   1. sudo modprobe industrialio kfifo_buf industrialio-sw-trigger
 *   2. sudo modprobe iio_dummy iio-trig-hrtimer
 *   3. sudo mkdir /configfs
 *   4. sudo mount -t configfs none /config
 *   5. sudo mkdir /config/iio/triggers/hrtimer/instance1
```

Step 4 fails because `/config` does not exist (should it be `/configfs`?) If I run step 4 with `/configfs` then step 5 fails, I think because we have already mounted and cannot modify the directory. I tried running step 5 before step 4 and that succeeds. However, I still get zero devices when running the dummy code.

Ideas?

I'm on a desktop running Linux Mint 18.2, and I'm on version 0.14 (tag 7dddeec) of the library.

mhennerich, 2018-04-12:
Step 3 should create a directory which can in step 4 mounted.
So 3 should be:
**sudo mkdir /config**

mhennerich, 2018-04-12:
Then make sure your mount succeeds.
Try typing mount afterwards...

mortenfyhn, 2018-04-12:
Thanks, that seems to help. I repeated 1 through 5 with 3 changed as you say (and still running 5 before 4).

With `mount` then I get
```
$ mount | grep 'config'
configfs on /sys/kernel/config type configfs (rw,relatime)
none on /config type configfs (rw,relatime)
```

The dummy example still doesn't run though.
```
$ ./dummy-iiostream
Library version: 0.14 (git tag: 7dddeec)
* Acquiring IIO context
dummy-iiostream: dummy-iiostream.c:190: main: Assertion `iio_context_get_devices_count(ctx) > 0 && "No devices"' failed.
[1]    28871 abort      ./dummy-iiostream
```

mhennerich, 2018-04-12:
configfs is already mounted to /sys/kernel/config!
So don't try to mount it twice. Skip step 3,4 
and juts do:
sudo mkdir /sys/kernel/config/iio/triggers/hrtimer/instance1

mortenfyhn, 2018-04-12:
Should I then unmount it first? `sudo mkdir -p /sys/kernel/config/iio/triggers/hrtimer/instance1` (I added the `-p`) gives a "Operation not permitted" message.

mhennerich, 2018-04-12:
I guess your kernel if build without CONFIG_IIO_CONFIGFS support !?

mortenfyhn, 2018-04-12:
Do you know how I can check?

mhennerich, 2018-04-12:
root@analog:~# cat /proc/config.gz | gzip -d | grep CONFIG_IIO_CONFIGFS
# CONFIG_IIO_CONFIGFS is not set
root@analog:~#

mortenfyhn, 2018-04-12:
I don't have a config.gz there, but I did find this:
```
$ cat /boot/config-4.10.0-30-generic | grep CONFIG_IIO_CONFIGFS
CONFIG_IIO_CONFIGFS=m
```

lclausen-adi, 2018-04-12:
When all the modules are loaded does /config/iio/triggers/hrtimer/ exist?
```ls /config/iio/triggers/hrtimer/```

mortenfyhn, 2018-04-12:
If I do
```
sudo modprobe industrialio kfifo_buf industrialio-sw-trigger
sudo modprobe iio_dummy iio-trig-hrtimer
sudo mkdir /config
sudo mount -t configfs none /config
sudo mkdir -p /config/iio/triggers/hrtimer/instance1
```
then the last line fails with "operation not permitted" (or no such file/dir if I skip the `-p`), and I have
```
$ tree /config           
/config
└── iio
    └── devices
        └── dummy

3 directories, 0 files
```

If I then `sudo umount /config`, I'm allowed to `sudo mkdir -p /config/iio/triggers/hrtimer/instance1`, and I get
```
/config
└── iio
    └── triggers
        └── hrtimer
            └── instance1

4 directories, 0 files
```
but when I `sudo mount -t configfs none /config` then `/config` reverts to `iio/devices/dummy`.

lclausen-adi, 2018-04-12:
Hm, so this is a bit strange. Usually when all the modules that you've loaded are loaded and the configfs is mounted there should be a directory called /config/iio/triggers/. Not sure what goes wrong. Can you try to change the order to see if there are any issues with that. Load industrialio-sw-trigger and iio-trig-hrtimer last after all other modules.

mortenfyhn, 2018-04-12:
I did a power cycle first this time as well.
`sudo rm -rf /config`
`sudo modprobe industrialio kfifo_buf iio_dummy` (`mount | grep 'config'` returns nothing at this point)
`sudo modprobe industrialio-sw-trigger iio-trig-hrtimer` (`mount | grep 'config'` now returns `configfs on /sys/kernel/config type configfs (rw,relatime)`)
`sudo mkdir /config`
`sudo mount -t configfs none /config`

At this point I get
```
$ tree /config
/config
└── iio
    └── triggers

2 directories, 0 files
```
which is something new. `sudo mkdir -p /config/iio/triggers/hrtimer/instance1` is still not permitted.

dbaluta, 2018-04-12:
Hi @mortenfyhn,

As @lclausen-adi  said it appears that the triggers directory is not created. So, lets try more debugging:

After doing:

```
$ sudo modprobe industrialio kfifo_buf industrialio-sw-trigger
$ sudo modprobe iio_dummy iio-trig-hrtimer
```

What is the output of lsmod  and dmesg commands? Use a service like pastebin.com to post the output.

Then, let us know the output of:

$ cat /proc/kallsyms | grep iio

Also, what is the output of mount command? I assume that configfs is already mounted on:

/sys/kernel/config

So, can you please let as know the output of tree /sys/kernel/config?

lclausen-adi, 2018-04-12:
Hm, this is really unexpected. There seems to be some kind of ordering issue where there shouldn't be one. Which kernel version are you using (```uname -a```)?

Maybe we can get to a working state by doing things step by step.
```
sudo modprobe industrialio
sudo modprobe industrialio-configfs
sudo modprobe industrialio-sw-device
sudo modprobe industrialio-sw-trigger
sudo modprobe iio-trig-hrtimer
sudo modprobe iio_dummy
sudo mkdir /config
sudo mount -t configfs none /config
```

mortenfyhn, 2018-04-12:
@dbaluta 
I ran the two modprobe commands and I get this: [lsmod](https://pastebin.com/9NNBfFzi), [dmesg](https://pastebin.com/6KGgyMYu), [`cat /proc/kallsyms | grep iio`](https://pastebin.com/9xndGpK6).
Furthermore:
```
$ mount | grep 'config'
configfs on /sys/kernel/config type configfs (rw,relatime)
none on /config type configfs (rw,relatime)

$ tree /sys/kernel/config
/sys/kernel/config
└── iio
    ├── devices
    │   └── dummy
    └── triggers

4 directories, 0 files

$ tree /config
/config
└── iio
    ├── devices
    │   └── dummy
    └── triggers

4 directories, 0 files
```

mortenfyhn, 2018-04-12:
@lclausen-adi 
```
$ uname -a
Linux D986 4.10.0-30-generic #34~16.04.1-Ubuntu SMP Wed Aug 2 02:13:56 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```

Will try modprobe in that order and update this reply.

dbaluta, 2018-04-12:
Hi @mortenfyhn,

Looking at dmesg:

[  890.213084] iio_dummy: unknown parameter 'iio-trig-hrtimer' ignored

I think the initial problem was the way modprobe is used. modprobe doesn't expect a list of modules to be loaded.

       modprobe [-v] [-V] [-C config-file] [-n] [-i] [-q] [-b] [modulename] [module parameters...]

daniel.

lclausen-adi, 2018-04-12:
@dbaluta That makes a lot of sense. Thanks :)

mortenfyhn, 2018-04-13:
Yup, I guess that was it. I'm still a bit confused about why the device count assertion succeeds (with count 1) but find device fails.

mortenfyhn, 2018-04-13:
Great! I shouldn't have missed that. Turns out that iio only finds the instance1 trigger device, not the actual dummy device.

lclausen-adi, 2018-04-13:
You might have to create it ```mkdir /config/iio/devices/dummy/my_dummy_device```

pcercuei, 2018-04-13:
I guess ```iio_context_find_device(ctx, name) ``` should set errno to ENODEV on error.

lclausen-adi, 2018-04-17:
We still need to fix the steps as documented in the file. I'd like to keep this open until that has been done.

markusr, 2018-04-26:
I had the same problem today. I am running Ubuntu 16.04 (x64). 
Here are steps I did:

1. Check if `configfs` is already mounted
```
$ mount | grep 'config'
configfs on /sys/kernel/config type configfs (rw,relatime)
```
It is mounted under `/sys/kernel/config`.

2. Load modules one by one
```
$ sudo modprobe industrialio
$ sudo modprobe industrialio-configfs
$ sudo modprobe industrialio-sw-device
$ sudo modprobe industrialio-sw-trigger
$ sudo modprobe iio-trig-hrtimer
$ sudo modprobe iio_dummy
```
3. Create trigger and dummy device under `/sys/kernel/config`
```
$ sudo mkdir /sys/kernel/config/iio/triggers/hrtimer/instance1
$ sudo mkdir /sys/kernel/config/iio/devices/dummy/my_dummy_device
```
4. Run `iio_info` to see that all worked properly
```
$ iio_info 
Library version: 0.14 (git tag: c9909f2)
Compiled with backends: local xml ip
IIO context created with local backend.
Backend version: 0.14 (git tag: c9909f2)
Backend description string: Linux markus-desktop 4.13.0-39-generic #44~16.04.1-Ubuntu SMP Thu Apr 5 16:43:10 UTC 2018 x86_64
IIO context has 1 attributes:
        local,kernel: 4.13.0-39-generic
IIO context has 2 devices:
        iio:device0: my_dummy_device
                10 channels found:
                        activity_walking:  (input)
                        1 channel-specific attributes found:
                                attr  0: input value: 4
...
```

lclausen-adi, 2018-05-02:
@markusr Do you mind if I put what you wrote as the instructions in the source file?

markusr, 2018-05-02:
@lclausen-adi Feel free to use it as you like.

lclausen-adi, 2018-05-02:
@markusr under what name/pseudonym do you want to be acknowledged?

markusr, 2018-05-03:
Just use my name: Markus Roppelt