analogdevicesinc/libiio #170: Errors in Phyton bindings

aadv17, 2018-05-06:
I installed libiio v0.14-19-gfa6ec67 on my Ubuntu 16.04 system.

When I try to run iio_info.py from libiio / bindings / python / examples I get this error:

AttributeError: /usr/lib/x86_64-linux-gnu/libiio.so.o: undefined symbol: iio_get_backends_count

Also iio.py in libiio / bindings / python shows these errors in the IDE:

Instance of '_Attr' has no '__read' member (367,32)

'Instance of '_Attr' has no '__write' member (367,32)

Undefined variable 'unicode' (735,51)

Class 'value' has no 'decode' member (745,16) (745,46)

lclausen-adi, 2018-05-07:
Hi,

Did you install from source, or did you install the package that comes with the discribtion?

If you installed from source make sure that there is no older version installed from the distribution.

- Lars

pcercuei, 2018-05-07:
What version of Python?

aadv17, 2018-05-07:
Originally I used a libiio package. However this time I installed libiio from source and the errors are gone.
(I now have a Ubuntu 18.04 system, which has Python 3.65 installed by default).
Thanks for help!

closed 2018-05-07 by lclausen-adi

ankostis, 2021-09-24:
Indeed, It seems that the `libiio` in debian repos contain an older version (<6mo) that does not contain the ` iio_get_backends_count()` function.