analogdevicesinc/libiio #173: Does pluto support arbitrary sample rate?

commthu, 2018-05-13:
I needs some specical sample rate, such as  6.5MHZ , 1.2288MHZ 15.36MHz, 17.333333MHz,  19.2MHz  etc.
doe pluto support arbitrary sample rate?

due to USB2.0 speed limit,   what is the maximum  sample rate  for stable receive (no data loss).

mhennerich, 2018-05-14:
Pluto related questions are better asked here: https://github.com/analogdevicesinc/plutosdr-fw

Yes pluto supports arbitrary baseband rates. With the additional decimation filter in the HDL design, it can go down to 65kSPS the max is 61.44MSPS. Frequency resolution is about 2.5Hz. You should be able to stream 6-7 MSPS continuously.

rgetz, 2018-05-21:
If you need anything else - let us know, otherwise closing

closed 2018-05-21 by rgetz