analogdevicesinc/libiio #177: building failed with: undefined reference to `libusb_set_auto_detach_kernel_driver'

drom, 2018-06-02:
When building from `master` on OpenSuSe Tumbleweed, I am getting the following error:

```sh
...
Scanning dependencies of target iio_info
[ 43%] Building C object tests/CMakeFiles/iio_info.dir/iio_info.c.o
[ 43%] Linking C executable iio_info
../libiio.so.0.15: undefined reference to `libusb_set_auto_detach_kernel_driver'
../libiio.so.0.15: undefined reference to `libusb_handle_events_completed'
collect2: error: ld returned 1 exit status
make[2]: *** [tests/CMakeFiles/iio_info.dir/build.make:85: tests/iio_info] Error 1
make[1]: *** [CMakeFiles/Makefile2:287: tests/CMakeFiles/iio_info.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

libusb-1_0-0  Version 1.0.22-1.1
kernel 4.16.12-1-default

commodo, 2018-06-04:
Hey,

Did you install also `libusb-1_0-0-devel` ?
I don't have a Tumbleweed at hand, but I do have a Leap.
For comparison, this setup works [ **zypper search libusb** ] :
```
S  | Name                   | Summary                                                             | Type  
---+------------------------+---------------------------------------------------------------------+--------
   | libhidapi-libusb0      | Simple library for communicating with USB and Bluetooth HID devices | package
i  | libusb-0_1-4           | libusb-1.0 Compatibility Library for libusb-0.1                     | package
   | libusb-0_1-4-32bit     | libusb-1.0 Compatibility Library for libusb-0.1                     | package
i  | libusb-1_0-0           | USB Library                                                         | package
   | libusb-1_0-0-32bit     | USB Library                                                         | package
i+ | libusb-1_0-devel       | USB Library                                                         | package
   | libusb-1_0-devel-32bit | USB Library                                                         | package
   | libusb-compat-devel    | libusb-1.0 Compatibility Layer for libusb-0.1                       | package
   | libusbg-devel          | Development files for the USB gadget configuration library          | package
   | libusbg-tools          | Utilities to show and configure USB gadget devices                  | package
   | libusbg0               | USB gadget device configuration library                             | package
   | libusbgx-devel         | Development files for the USB gadget configuration library          | package
   | libusbgx-tools         | Utilities to show and configure USB gadget devices                  | package
   | libusbgx1              | USB gadget device configuration library                             | package
   | libusbmuxd-devel       | Development files for libusbmuxd                                    | package
i  | libusbmuxd4            | A client library to multiplex connections from and to iOS devices   | package
   | libusbmuxd4-32bit      | A client library to multiplex connections from and to iOS devices   | package
   | libusbprog0            | USBprog Library                                                     | package
   | libusbredirhost1       | A protocol for redirection USB traffic                              | package
   | libusbredirparser1     | A protocol for redirection USB traffic                              | package
```

commodo, 2018-06-04:
[curious] why are you using `libusb-1_0-0-32bit` version ? i wonder if that came with Tumbleweed, or some other package installed it as a dependency;

you might have a multi-arch issue, between the 32bit version of the lib and the compiler that may compile this for 64bit [multi-arch can be pretty annoying, and tricky to handle]

maybe first try to remove `libusb-1_0-0-32bit` if it is possible [make sure that files for 32bit are gone], and if that doesn't work, try installing `libusb-0_1-4`

commodo, 2018-06-05:
did you clean the build before re-building ;
i'm not sure how cmake caches stuff; 

i wonder how your libusb libs look like;
on my Leap, they look like this:
```
/usr/include/libusb-1.0
/usr/include/libusb-1.0/libusb.h
/usr/lib64/pkgconfig/libusb-1.0.pc
/usr/lib64/libusb-1.0.so.0
/usr/lib64/libusb-1.0.so.0.1.0
/usr/lib64/libusb-0.1.so.4
/usr/lib64/libusb-0.1.so.4.4.4
/usr/lib64/libusb-1.0.so
```

drom, 2018-06-05:
Yes, I deleted build folder and rerun cmake every time.

```
-rw-r--r-- 1 root root 72802 26 mars  03:28 /usr/include/libusb-1.0/libusb.h

lrwxrwxrwx 1 root root     19 11 janv. 01:10 /usr/lib64/libusb-0.1.so.4 -> libusb-0.1.so.4.4.4
-rwxr-xr-x 1 root root  22656 11 janv. 01:10 /usr/lib64/libusb-0.1.so.4.4.4
lrwxrwxrwx 1 root root     19 26 mars  03:28 /usr/lib64/libusb-1.0.so -> libusb-1.0.so.0.1.0
lrwxrwxrwx 1 root root     19 26 mars  03:28 /usr/lib64/libusb-1.0.so.0 -> libusb-1.0.so.0.1.0
-rwxr-xr-x 1 root root 101384 26 mars  03:28 /usr/lib64/libusb-1.0.so.0.1.0
lrwxrwxrwx 1 root root     19 11 janv. 04:53 /usr/lib64/libusbmuxd.so.4 -> libusbmuxd.so.4.0.0
-rwxr-xr-x 1 root root  31216 11 janv. 04:53 /usr/lib64/libusbmuxd.so.4.0.0
```

pcercuei, 2018-06-06:
Could you compile with CMAKE_VERBOSE_MAKEFILE? I just want to be sure that it tries to link with libusb-1.0 and not libusb-0.1 which does not have these symbols.

drom, 2018-06-07:
Yes, sure. Here it is:
```
[ 43%] Building C object tests/CMakeFiles/iio_info.dir/iio_info.c.o
cd /home/chjulia/work/github/analogdevicesinc/libiio/build/tests && /usr/bin/cc -DLIBIIO_EXPORTS=1 -DLOCAL_BACKEND=1 -DMATLAB_BINDINGS_API=1 -DNETWORK_BACKEND=1 -DUSB_BACKEND=1 -DXML_BACKEND=1 -D_GNU_SOURCE=1 -D_POSIX_C_SOURCE=200809L -D__XSI_VISIBLE=500 -I/home/chjulia/work/github/analogdevicesinc/libiio -I/home/chjulia/work/github/analogdevicesinc/libiio/build -I/usr/include/libxml2  -fvisibility=hidden -Wall -Wextra -Wno-unused-parameter -Wno-sign-compare -Wpedantic -O2 -g -DNDEBUG   -std=c99 -o CMakeFiles/iio_info.dir/iio_info.c.o   -c /home/chjulia/work/github/analogdevicesinc/libiio/tests/iio_info.c
[ 45%] Linking C executable iio_info
cd /home/chjulia/work/github/analogdevicesinc/libiio/build/tests && /usr/bin/cmake -E cmake_link_script CMakeFiles/iio_info.dir/link.txt --verbose=1
/usr/bin/cc  -fvisibility=hidden -Wall -Wextra -Wno-unused-parameter -Wno-sign-compare -Wpedantic -O2 -g -DNDEBUG  -rdynamic CMakeFiles/iio_info.dir/iio_info.c.o  -o iio_info -Wl,-rpath,/home/chjulia/work/github/analogdevicesinc/libiio/build: ../libiio.so.0.15 
../libiio.so.0.15: undefined reference to `libusb_set_auto_detach_kernel_driver'
../libiio.so.0.15: undefined reference to `libusb_handle_events_completed'
collect2: error: ld returned 1 exit status
make[2]: *** [tests/CMakeFiles/iio_info.dir/build.make:88: tests/iio_info] Error 1
make[2]: Leaving directory '/home/chjulia/work/github/analogdevicesinc/libiio/build'
make[1]: *** [CMakeFiles/Makefile2:290: tests/CMakeFiles/iio_info.dir/all] Error 2
make[1]: Leaving directory '/home/chjulia/work/github/analogdevicesinc/libiio/build'
make: *** [Makefile:133: all] Error 2
```

pcercuei, 2018-06-07:
I will need a bit more log, when it links the libiio itself.

drom, 2018-06-08:
This is full log.
```
chjulia@carbon:~/work/github/analogdevicesinc/libiio/build> make
/usr/bin/cmake -H/home/chjulia/work/github/analogdevicesinc/libiio -B/home/chjulia/work/github/analogdevicesinc/libiio/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/chjulia/work/github/analogdevicesinc/libiio/build/CMakeFiles /home/chjulia/work/github/analogdevicesinc/libiio/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/chjulia/work/github/analogdevicesinc/libiio/build'
make -f CMakeFiles/iio.dir/build.make CMakeFiles/iio.dir/depend
make[2]: Entering directory '/home/chjulia/work/github/analogdevicesinc/libiio/build'
cd /home/chjulia/work/github/analogdevicesinc/libiio/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/chjulia/work/github/analogdevicesinc/libiio /home/chjulia/work/github/analogdevicesinc/libiio /home/chjulia/work/github/analogdevicesinc/libiio/build /home/chjulia/work/github/analogdevicesinc/libiio/build /home/chjulia/work/github/analogdevicesinc/libiio/build/CMakeFiles/iio.dir/DependInfo.cmake --color=
Scanning dependencies of target iio
make[2]: Leaving directory '/home/chjulia/work/github/analogdevicesinc/libiio/build'
make -f CMakeFiles/iio.dir/build.make CMakeFiles/iio.dir/build
make[2]: Entering directory '/home/chjulia/work/github/analogdevicesinc/libiio/build'
[  2%] Building C object CMakeFiles/iio.dir/backend.c.o
/usr/bin/cc -DLIBIIO_EXPORTS=1 -DLOCAL_BACKEND=1 -DMATLAB_BINDINGS_API=1 -DNETWORK_BACKEND=1 -DUSB_BACKEND=1 -DXML_BACKEND=1 -D_GNU_SOURCE=1 -D_POSIX_C_SOURCE=200809L -D__XSI_VISIBLE=500 -Diio_EXPORTS -I/home/chjulia/work/github/analogdevicesinc/libiio -I/home/chjulia/work/github/analogdevicesinc/libiio/build -I/usr/include/libxml2  -fvisibility=hidden -Wall -Wextra -Wno-unused-parameter -Wno-sign-compare -Wpedantic -O2 -g -DNDEBUG -fPIC   -std=c99 -o CMakeFiles/iio.dir/backend.c.o   -c /home/chjulia/work/github/analogdevicesinc/libiio/backend.c
[  5%] Building C object CMakeFiles/iio.dir/channel.c.o
/usr/bin/cc -DLIBIIO_EXPORTS=1 -DLOCAL_BACKEND=1 -DMATLAB_BINDINGS_API=1 -DNETWORK_BACKEND=1 -DUSB_BACKEND=1 -DXML_BACKEND=1 -D_GNU_SOURCE=1 -D_POSIX_C_SOURCE=200809L -D__XSI_VISIBLE=500 -Diio_EXPORTS -I/home/chjulia/work/github/analogdevicesinc/libiio -I/home/chjulia/work/github/analogdevicesinc/libiio/build -I/usr/include/libxml2  -fvisibility=hidden -Wall -Wextra -Wno-unused-parameter -Wno-sign-compare -Wpedantic -O2 -g -DNDEBUG -fPIC   -std=c99 -o CMakeFiles/iio.dir/channel.c.o   -c /home/chjulia/work/github/analogdevicesinc/libiio/channel.c
[  8%] Building C object CMakeFiles/iio.dir/device.c.o
/usr/bin/cc -DLIBIIO_EXPORTS=1 -DLOCAL_BACKEND=1 -DMATLAB_BINDINGS_API=1 -DNETWORK_BACKEND=1 -DUSB_BACKEND=1 -DXML_BACKEND=1 -D_GNU_SOURCE=1 -D_POSIX_C_SOURCE=200809L -D__XSI_VISIBLE=500 -Diio_EXPORTS -I/home/chjulia/work/github/analogdevicesinc/libiio -I/home/chjulia/work/github/analogdevicesinc/libiio/build -I/usr/include/libxml2  -fvisibility=hidden -Wall -Wextra -Wno-unused-parameter -Wno-sign-compare -Wpedantic -O2 -g -DNDEBUG -fPIC   -std=c99 -o CMakeFiles/iio.dir/device.c.o   -c /home/chjulia/work/github/analogdevicesinc/libiio/device.c
[ 10%] Building C object CMakeFiles/iio.dir/context.c.o
/usr/bin/cc -DLIBIIO_EXPORTS=1 -DLOCAL_BACKEND=1 -DMATLAB_BINDINGS_API=1 -DNETWORK_BACKEND=1 -DUSB_BACKEND=1 -DXML_BACKEND=1 -D_GNU_SOURCE=1 -D_POSIX_C_SOURCE=200809L -D__XSI_VISIBLE=500 -Diio_EXPORTS -I/home/chjulia/work/github/analogdevicesinc/libiio -I/home/chjulia/work/github/analogdevicesinc/libiio/build -I/usr/include/libxml2  -fvisibility=hidden -Wall -Wextra -Wno-unused-parameter -Wno-sign-compare -Wpedantic -O2 -g -DNDEBUG -fPIC   -std=c99 -o CMakeFiles/iio.dir/context.c.o   -c /home/chjulia/work/github/analogdevicesinc/libiio/context.c
[ 13%] Building C object CMakeFiles/iio.dir/buffer.c.o
/usr/bin/cc -DLIBIIO_EXPORTS=1 -DLOCAL_BACKEND=1 -DMATLAB_BINDINGS_API=1 -DNETWORK_BACKEND=1 -DUSB_BACKEND=1 -DXML_BACKEND=1 -D_GNU_SOURCE=1 -D_POSIX_C_SOURCE=200809L -D__XSI_VISIBLE=500 -Diio_EXPORTS -I/home/chjulia/work/github/analogdevicesinc/libiio -I/home/chjulia/work/github/analogdevicesinc/libiio/build -I/usr/include/libxml2  -fvisibility=hidden -Wall -Wextra -Wno-unused-parameter -Wno-sign-compare -Wpedantic -O2 -g -DNDEBUG -fPIC   -std=c99 -o CMakeFiles/iio.dir/buffer.c.o   -c /home/chjulia/work/github/analogdevicesinc/libiio/buffer.c
[ 16%] Building C object CMakeFiles/iio.dir/utilities.c.o
/usr/bin/cc -DLIBIIO_EXPORTS=1 -DLOCAL_BACKEND=1 -DMATLAB_BINDINGS_API=1 -DNETWORK_BACKEND=1 -DUSB_BACKEND=1 -DXML_BACKEND=1 -D_GNU_SOURCE=1 -D_POSIX_C_SOURCE=200809L -D__XSI_VISIBLE=500 -Diio_EXPORTS -I/home/chjulia/work/github/analogdevicesinc/libiio -I/home/chjulia/work/github/analogdevicesinc/libiio/build -I/usr/include/libxml2  -fvisibility=hidden -Wall -Wextra -Wno-unused-parameter -Wno-sign-compare -Wpedantic -O2 -g -DNDEBUG -fPIC   -std=c99 -o CMakeFiles/iio.dir/utilities.c.o   -c /home/chjulia/work/github/analogdevicesinc/libiio/utilities.c
[ 18%] Building C object CMakeFiles/iio.dir/scan.c.o
/usr/bin/cc -DLIBIIO_EXPORTS=1 -DLOCAL_BACKEND=1 -DMATLAB_BINDINGS_API=1 -DNETWORK_BACKEND=1 -DUSB_BACKEND=1 -DXML_BACKEND=1 -D_GNU_SOURCE=1 -D_POSIX_C_SOURCE=200809L -D__XSI_VISIBLE=500 -Diio_EXPORTS -I/home/chjulia/work/github/analogdevicesinc/libiio -I/home/chjulia/work/github/analogdevicesinc/libiio/build -I/usr/include/libxml2  -fvisibility=hidden -Wall -Wextra -Wno-unused-parame [... cut ...]
../libiio.so.0.15: undefined reference to `libusb_set_auto_detach_kernel_driver'
../libiio.so.0.15: undefined reference to `libusb_handle_events_completed'
collect2: error: ld returned 1 exit status
make[2]: *** [tests/CMakeFiles/iio_info.dir/build.make:88: tests/iio_info] Error 1
make[2]: Leaving directory '/home/chjulia/work/github/analogdevicesinc/libiio/build'
make[1]: *** [CMakeFiles/Makefile2:290: tests/CMakeFiles/iio_info.dir/all] Error 2
make[1]: Leaving directory '/home/chjulia/work/github/analogdevicesinc/libiio/build'
make: *** [Makefile:133: all] Error 2
```

pcercuei, 2018-06-08:
Ok, so it does link to libusb-1.0.

drom, 2018-06-09:
Yes, it fails when making `iio_info` it seems.

pcercuei, 2018-06-09:
Well, if libiio links with a libusb that is missing symbols, it will still link fine and these symbols are marked as unresolved (for the case where they belong to another shared library). So the fact that it links does not mean it does it properly.

Could you run ```nm /usr/lib64/libusb-1.0.so |grep libusb_set_auto_detach_kernel_driver``` and see what it returns?

And also: ```ldd libiio.so.0.15``` in your build folder.

drom, 2018-06-10:
`nm: /usr/lib64/libusb-1.0.so: no symbols`

drom, 2018-06-10:
```
ldd libiio.so.0.15 
        linux-vdso.so.1 (0x00007ffc5abca000)
        librt.so.1 => /lib64/librt.so.1 (0x00007f561d94b000)
        libusb-1.0.so.0 => /usr/local/lib/libusb-1.0.so.0 (0x00007f561d73d000)
        libxml2.so.2 => /usr/lib64/libxml2.so.2 (0x00007f561d3d5000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f561d1b6000)
        liblzma.so.5 => /usr/lib64/liblzma.so.5 (0x00007f561cf7b000)
        libz.so.1 => /lib64/libz.so.1 (0x00007f561cd64000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f561c9d1000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f561c613000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007f561c40f000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f561dd6d000)
```

pcercuei, 2018-06-10:
There's the problem. You have a local libusb installed in /usr/local, which appears to be older. Remove it (it shouldn't be needed, right?) and then libiio should build fine.

drom, 2018-06-10:
Yes, thank you. 
Looks like I was building `libusb` long time ago (2013) and totally forgot about it :(
My fault.

closed 2018-06-10 by drom