analogdevicesinc/libiio #178: Add a function to synchronize an IIO context

allsey87, 2018-06-25:
During running an application it is possible that IIO devices appear or disappear. In my particular case, I would like to be able to add a software trigger (a high resolution timer trigger) for each instance of a given IIO device type and then assign those triggers through libiio.

I suggest at a minimum implementing a `iio_context_sync(struct iio_context* ctx)` function that attempts to add devices that were not present at the time of the innovation of `iio_create_local_context`. This way, when I call `mkdir("/sys/kernel/config/iio/triggers/hrtimer/newtrigger, S_IRWXU | S_IRWXG | S_IRWXO)`, I can access the new IIO device from the existing context. The difference between calling the suggested function and just destroying and recreating the context would be that `iio_context_sync(struct iio_context* ctx)` doesn't invalidate pointers to devices, channels etc.

Alternatively, it would be even nicer to have some functions that automatically set up commonly used software triggers such as sysfs and hrtimer, and just directly return a new iio_device pointer on success.

rgetz, 2018-07-04:
@allsey87 : can you describe your use case in a little more detail? How do IIO devices appear, and in what context (local or remote) do you want to reference things to?

allsey87, 2018-07-04:
@rgetz for my use case, IIO triggers (devices) appear when they are created by my program. For example, hrtimer triggers are IIO devices which I create by calling mkdir inside my program to create the appropriate directory inside of configfs.

The software that I am writing needs to create a hrtimer trigger for each IIO device found that matches a given name. The problem is once I have created my local context and loop over the devices, I would need to destroy that local context and recreate it in order to use the newly created trigger IIO devices inside of libiio. I believe the ability to sync a context would solve this problem.

Another use case for the ability to sync a context would be for IIO devices that are added when USB hardware is attached. For example, a udev rule could be configured to send a signal to a running program that IIO devices have been added or removed and that the local context needs to be synchronised with sysfs.

rgetz, 2018-10-31:
@allsey87 : I think that destroying things, and re-create it is the "proper" way right now. There might be issues (some pointers might not be valid) anymore if devices inserted themselves in a current context, due to requirements for sorting devices/channels in certain ways.

@pcercuei might be able to comment more - but I think this would have to be a version 2.0 enhancement.

-Robin

rgetz, 2018-11-12:
Closing this as can't/won't fix.

closed 2018-11-12 by rgetz