analogdevicesinc/libiio #179: Troubles with installing all components of iio for gnu radio ; both mac and linux

glangsto, 2018-06-26:
Hello
This error concerns one aspect of installing libiio for use with Pluto SDRs.
I've had considerable (months) of trouble getting this to work, despite
having good luck with many other packages.   I've been able to run
gnu radio on Macs and many versions of linux.

However my goal is to get PlutoSDR to work with gnuradio-companion.  This
requires 

gr-iio 
Using git clone 
I'm currently using a odroid-xu4 which is great for most packages.
However the gr-iio "make" step for is crashing the system, forcing a reboot.
I think this is because the build is utilizing all 2GB of memory (or something).
This is inside the compile stage.  Here are the messages:

==========================================
[ 61%] Linking CXX executable iio_pluto_source_swig_swig_2d0df
Swig source
Swig source
Swig source
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
lib/CMakeFiles/gnuradio-iio.dir/build.make:313: recipe for target 'lib/CMakeFiles/gnuradio-iio.dir/iio_math_gen_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-iio.dir/iio_math_gen_impl.cc.o] Error 4
make[2]: *** Waiting for unfinished jobs....
[ 61%] Built target iio_pluto_source_swig_swig_2d0df
[ 61%] Built target iio_pluto_sink_swig_swig_2d0df
Scanning dependencies of target pygen_swig_dd92f
Scanning dependencies of target pygen_swig_6d49c
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
lib/CMakeFiles/gnuradio-iio.dir/build.make:169: recipe for target 'lib/CMakeFiles/gnuradio-iio.dir/fmcomms5_source_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-iio.dir/fmcomms5_source_impl.cc.o] Error 4
[ 68%] Generating iio_pluto_sink_swig.pyc
[ 68%] Generating iio_pluto_source_swig.pyo
[ 68%] Generating iio_pluto_sink_swig.pyo
[ 68%] Generating iio_pluto_source_swig.pyc
...
=================================
Eventually the system just hangs, forcing a hard reboot.

Is there a way to reduce the memory usage inside of the make for this library?
Perhaps there is a way to divide the process?   
Please redirect to the correct developer, if this is the wrong forum.

Best regards

Glen

pcercuei, 2018-06-26:
It gets killed because the system runs out of RAM, yes...
Try to compile with just one thread (make -j1), and/or enable swap, that will be extremely slow but it should still work.

rgetz, 2018-07-22:
This likely has to do more with https://github.com/analogdevicesinc/gr-iio than with libiio, so unless @glangsto has an update - closing this. Feel free to open a new issue in the gr-iio repo if you are still having issues.

closed 2018-07-22 by rgetz