analogdevicesinc/libiio #180: incorrect output rate for iio_readdev

f--t, 2018-07-19:
Reading a single channel of cf-ad9361-lpc produces almost twice of the output as it should. For examples, with sampling at 4MSPS, one channel -- voltage0 -- should not produce more than 8 MiB/s but iio_readdev reads almost twice that amount. However, reading two channels -- voltage0, voltage1 -- results in an approximately correct rate of 16 MiB/s.

```
        iio:device4: cf-ad9361-lpc (buffer capable)
                2 channels found:
                        voltage0:  (input, index: 0, format: le:S12/16>>0)
                        6 channel-specific attributes found:
                                attr  0: calibphase value: 0.000000
                                attr  1: calibbias value: 0
                                attr  2: calibscale value: 1.000000
                                attr  3: samples_pps ERROR: No such device (-19)
                                attr  4: sampling_frequency_available value: 3999999 499999
                                attr  5: sampling_frequency value: 3999999
                        voltage1:  (input, index: 1, format: le:S12/16>>0)
                        6 channel-specific attributes found:
                                attr  0: calibbias value: 0
                                attr  1: calibphase value: 0.000000
                                attr  2: calibscale value: 1.000000
                                attr  3: samples_pps ERROR: No such device (-19)
                                attr  4: sampling_frequency_available value: 3999999 499999
                                attr  5: sampling_frequency value: 3999999
```
```(bash)
$ iio_readdev -b 8192 -u ip:192.168.2.1  cf-ad9361-lpc voltage0 | pv -a > /dev/null
[14.0MiB/s]

$ iio_readdev -b 8192 -u ip:192.168.2.1  cf-ad9361-lpc voltage0 voltage1 | pv -a > /dev/null
[15.2MiB/s]
```

mhennerich, 2018-07-19:
Hi,

Thanks for reporting. It's not an issue of iio_readdev. Could it be that you're using a PlutoSDR?
On Pluto there is no PACK HDL core, it always sends both I&Q.

-Michael

f--t, 2018-07-19:
Thanks for the reply Michael,
Yes, I'm indeed using the PlutoSDR!

closed 2018-07-19 by f--t