analogdevicesinc/libiio #181: Keeping the APT package(s) up-to-date

pietos01, 2018-07-20:
On Ubuntu 16.04 LTS, I see that there exist APT packages for `libiio` and its Python bindings. Unfortunately, they seem to be stuck at version 0.6.40-1 which is years old. Furthermore, the README in this repository does not mention installing `libiio` through APT at all even though a .deb package is provided.

Is there a reason for not keeping these APT packages up-to-date (in particular `libiio0`)? Could we imagine these being supported in the future?

lclausen-adi, 2018-07-20:
Hi,

The apt packages are kept up to date by upstream Debian. See https://packages.debian.org/sid/libiio0

It is just that once a version of Ubuntu or Debian has been released with a particular version of a package they do not get upgraded within that version. Going to the next operating system release will get you a newer version of the package. This is not libiio specific, but the general approach used by distributions for packaging.

pietos01, 2018-07-20:
Thank you for your answer. It looks like I should have done a bit more research beforehand!

So if one wants to install the latest version of `libiio`, the "best way" is to download (`wget`, ...) the latest .deb and `apt install` it, right?

Now, about the Python wrapper, I see that the .deb extracts the files to `/usr/lib/python2.7/site-packages/` which may not be where the Python interpreter looks for `import`s (*e.g.* when using a `conda` distribution). Is there a way to easily install this wrapper in the "right location" by relying on `pip` (or other)?

As far as I am aware, the best way I found is through
```bash
cp /usr/lib/python2.7/site-packages/iio.py [...]
```

Although this may be overkill for a single file, could we image having a `libiio` (or `iio`) package on PyPI to make the whole distribution+installation process strong and trivial? Or is there a better I can't think of?

lclausen-adi, 2018-07-20:
The Debian packages install the python bindings in the location where the Debian distribution expects them.

If your setup has other requirements it might be best to install from source so you can choose the installation path.

closed 2018-07-20 by pietos01