analogdevicesinc/libiio #185: Wiki: Discuss the assignment of triggers when creating buffers

allsey87, 2018-07-27:
I suggest that https://analogdevicesinc.github.io/libiio/#create_buffer be updated to mention that a trigger (at least in the case of a triggered buffer) needs to be assigned *before* the buffer can be created. Otherwise the call to `local_write_dev_attr(dev, "buffer/enable", "1", 2, false);` in local.c fails with -22 (EINVAL).

I would also modify `static ssize_t local_enable_buffer(const struct iio_device *dev)` to print an warning/error when local_write_dev_attr fails. For the moment, no error is reported which made this issue difficult to find.

pcercuei, 2018-08-13:
You're right, the error is ignored and that's not normal.

rgetz, 2018-10-30:
@allsey87 : I have a patch for master #213 - can you try it out, and see if that would have helped in debugging things?

rgetz, 2018-11-01:
patch was applied, so closing this.

closed 2018-11-01 by rgetz