analogdevicesinc/libiio #186: Corrupted Memory using iio_scan_context_get_info_list()

hiderplo, 2018-08-13:
I am attempting to scan for a connected device (specifically the ADALM-PLUTO) which I know can be found by using uri "usb:1.3.5".  The purpose for this is to discover the device if it's address changes to something like "usb:1.5.5".  I cannot 'hardcode' a uri that might change from hardware to hardware.

I have discovered that using the following code, I can attain a valid uri for the device:
```c
#include <iio.h>
#include <stdio.h>

int main (int argc, char** argv) {
	iio_scan_context* scn = iio_create_scan_context(NULL, 0);
	iio_context_info*** lst;
	ssize_t n = iio_scan_context_get_info_list(scn, lst);
	printf("%s\n", iio_context_info_get_uri(lst[0][1]));
	iio_scan_context_destroy(scn);
	iio_context_info_list_free(*lst);
	return 0;
}
```
What if I want to create a list of all the valid URI's to try though.  I need to loop through this list and grab each one.  Adding even a simple looping structure results in a segmentation fault occurring inside the iio_scan_context_get_info_list(scn, lst); function.  To reproduce this, run the following:
```c
#include <iio.h>
#include <stdio.h>

int main (int argc, char** argv) {
	iio_scan_context* scn = iio_create_scan_context(NULL, 0);
	iio_context_info*** lst;
	ssize_t n = iio_scan_context_get_info_list(scn, lst);
	for(int i = 0; i < 1; i++) {}
	printf("%s\n", iio_context_info_get_uri(lst[0][1]));
	iio_scan_context_destroy(scn);
	iio_context_info_list_free(*lst);
	return 0;
}
```

Note that the only change to the two blocks of code are the addition of a `for(int i = 0; i < 1; i++) {}` line.  The only thing this line does is initialize an int, increment it once, then continue.  This loop is never run though, after adding some print statements you can isolate the seg fault down to the `ssize_t n = iio_scan_context_get_info_list(scn, lst);` line.

When running these snippets I used `g++ -o runText (src) -liio`

<details><summary>Valgrind Results for BAD snippet:</summary>
<p>

```
$ valgrind --tool=memcheck --leak-check=yes --show-reachable=yes --num-callers=20 --track-fds=yes ./runTest
==86016== Memcheck, a memory error detector
==86016== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==86016== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==86016== Command: ./runTest
==86016== 
==86016== Use of uninitialised value of size 8
==86016==    at 0x4E409DF: iio_scan_context_get_info_list (scan.c:71)
==86016==    by 0x400811: main (in /home/redhawk/Desktop/runTest)
==86016== 
==86016== 
==86016== Process terminating with default action of signal 11 (SIGSEGV)
==86016==  Bad permissions for mapped region at address 0x4006F0
==86016==    at 0x4E409DF: iio_scan_context_get_info_list (scan.c:71)
==86016==    by 0x400811: main (in /home/redhawk/Desktop/runTest)
==86016== 
==86016== FILE DESCRIPTORS: 9 open at exit.
==86016== Open file descriptor 8:
==86016==    at 0x596F637: timerfd_create (in /usr/lib64/libc-2.17.so)
==86016==    by 0x5E4FBBE: ??? (in /usr/lib64/libusb-1.0.so.0.1.0)
==86016==    by 0x5E4B25F: libusb_init (in /usr/lib64/libusb-1.0.so.0.1.0)
==86016==    by 0x4E45CA7: usb_context_scan_init (usb.c:1119)
==86016==    by 0x4E40B0B: iio_create_scan_context (scan.c:147)
==86016==    by 0x4007FA: main (in /home/redhawk/Desktop/runTest)
==86016== 
==86016== Open file descriptor 7:
==86016==    at 0x5960407: pipe (in /usr/lib64/libc-2.17.so)
==86016==    by 0x5E5060B: ??? (in /usr/lib64/libusb-1.0.so.0.1.0)
[... 674 lines cut ...]
$ cat /proc/version
Linux version 3.10.0-862.9.1.el7.x86_64 (builder@kbuilder.dev.centos.org) (gcc version 4.8.5 20150623 (Red Hat 4.8.5-28) (GCC) ) #1 SMP Mon Jul 16 16:29:36 UTC 2018
```

```
$ yum info libiio
...
Installed Packages
Name        : libiio
Arch        : x86_64
Version     : 0.15.g6ecff5d
Release     : 1
Size        : 1.6 M
Repo        : installed
Summary     : libiio built using CMake
License     : unknown
Description : DESCRIPTION
            : ===========
            : 
            : This is an installer created using CPack (http://www.cmake.org).
            : No additional installation instructions provided.
```

</p>
</details>

lclausen-adi, 2018-08-13:
Hi try
```
struct iio_context_info** lst;
...
n = iio_scan_context_get_info_list(ctx, &lst);
...
```

lclausen-adi, 2018-08-13:
It works because that is how it is supposed to be used :wink:

iio_scan_context_get_info_list() expects a valid memory location where it can store the result.

What you passed was an uninitialized pointer, this means a random memory location. If you are lucky like in your first example that random memory location was valid memory and it did not crash. In your second example that random location was invalid memory and it crashed.

closed 2018-08-13 by lclausen-adi

hiderplo, 2018-08-13:
That makes a whole lot of sense and I'm disappointed in myself for not putting it together.  Thank you for the feedback, it is greatly appreciated!