analogdevicesinc/libiio #188: iiod crashes, freezing target device - usb context

djcopley, 2018-08-15:
After closing a (usb) connection to iiod, target device becomes completely unresponsive, requiring a forced reboot.

lclausen-adi, 2018-08-16:
Hi,

What system are you running usbd on? In the past we've seen issues on certain platforms due to buggy USB drivers.

djcopley, 2018-08-16:
Its an ad9361 coupled with a zync ultrascale+ fpga. I have tried both the 4.9 and 4.14 versions of the kernel.

lclausen-adi, 2018-08-16:
Yes io_cancel() can be an issue. Also see https://ez.analog.com/message/359769-re-iiod-usb30-superspeed-doesn-t-work

We've made some fixes for the worst issues, but I guess there is still some more work to do.

You are using the kernel from the ADI github right?

djcopley, 2018-08-16:
Yes I am using the ADI Kernel. I believe its the latest version but will check to confirm.

As far as circumstances leading up to the failure, the crash always occurs when a connection is closed, not while serving data. It doesn't however happen every time a connection is closed, but it always happens eventually. I haven't counted but I would say it crashes around every fifth run or so (streaming with gnuradio, iio_readdev doesn't seem to crash iiod as frequently).

lclausen-adi, 2018-08-16:
Yes, that is consistent with the other report.

The issue seems to be cancelling the URB. Closing the connection is one of the paths that can trigger cancelling an URB (There are others like closing IIOD).

rgetz, 2020-01-13:
@djcopley : Do you remember if you were doing other IIO things in different threads? I have a single threaded application that is just doing
```
ctx = iio_create_context_from_uri(arg_uri);
iio_context_destroy(ctx);
```

and I can run it for 10 min (7424 times of open/close) on usb context without causing a problem (assuming that "good" is a non-NULL is returned from `iio_create_context_from_uri`). But there is no buffer, no channel access - nothing in between, simple open()/close(), and wonder if I'm missing something that could cause a problem.

@pcercuei or @mhennerich : if you have any comments - let me know

rgetz, 2020-01-13:
60m0.185s , ran 44,517 times successfully. I will do an overnight test, and update in the morning.

-Robin

mhennerich, 2020-01-14:
The original issue was only for MPSoC's DWC3 controller. Lars fixed a number related BUGs.
Since then this issue has no longer been seen!

Closing this - feel free to reopen if the issue appears in the 4.19 kernel we updated our master too.

closed 2020-01-14 by mhennerich

rgetz, 2020-01-14:
Ok - thanks - I tried the simple open/close last night (12 hours) successfully ran 534,152 times (on Pluto). I will try on the ADRV9009 SOM later and see if anything exists, and re-open if necessary