analogdevicesinc/libiio #190: iio_buffer_push_partial() is broken

aholtzma, 2018-09-07:
As of commit 19c1f6decdf28058bdbd0211b08ed0c9052661ed, calls to iio_buffer_push_partial() will always return -EBIG. This commit fixed one problem, but introduced a new problem later on here: 
```
	/* Requested buffer size is too big! */
	if (pdata->last_dequeued < 0 && bytes_used != block.size)
		return -EFBIG;
```
In the case of iio_buffer_push_partial(), bytes_used != block.size and will get erroneously caught here. 

It's not clear to me what this check is actually doing. Can someone comment on the rationale here?

pcercuei, 2018-11-06:
Sorry for the delay. The ```bytes_used != block.size``` expression is used to verify that the kernel does not return a buffer that is smaller than the number of bytes that are requested. If you call ```iio_buffer_push_partial``` right after creating the buffer with a small number of bytes, the call will fail, which is indeed a bug. I think it should be changed to ```bytes_used > block.size```.

mndza, 2018-12-03:
I am also using an old version of libiio due to this bug.
Tried the `bytes_used > block.size` fix and it seems to work OK.

rgetz, 2020-02-04:
@dNechita has an open pull request for this issue, so closing it here.

commit 29146750 (2020-02-04): local: Fix the checking of the buffer size returned by kernel

Fixes #190
closed 2020-02-04 by rgetz