analogdevicesinc/libiio #193: yum install issue on Centos 7.5

wrthissell, 2018-09-19:
Dear ADI engineers,
      I participated in the GNU Radio Conference workshop today on the Pluto SDR.  I tried to install the below referenced rpms on a Centos 7.5 Lenovo Thinkpad P50 and the rpm(s) is (are) requiring a libiio to be already installed.  I suspect that this is an issue.  See below:

[wrthissell@localhost x86_64]$ sudo yum install libiio-0.15.g6ecff5d-Linux-trusty.rpm libiio-0.15.g6ecff5d-Linux-precise.rpm
Loaded plugins: fastestmirror, langpacks, nvidia
Examining libiio-0.15.g6ecff5d-Linux-trusty.rpm: libiio-0.15.g6ecff5d-1.x86_64
Marking libiio-0.15.g6ecff5d-Linux-trusty.rpm to be installed
Examining libiio-0.15.g6ecff5d-Linux-precise.rpm: libiio-0.15.g6ecff5d-1.x86_64
Marking libiio-0.15.g6ecff5d-Linux-precise.rpm to be installed
Resolving Dependencies
--> Running transaction check
---> Package libiio.x86_64 0:0.15.g6ecff5d-1 will be installed
--> Processing Dependency: libiio.so.0()(64bit) for package: libiio-0.15.g6ecff5d-1.x86_64
Loading mirror speeds from cached hostfile
 * base: mirrors.xtom.com
 * centosplus: mirror.us-midwest-1.nexcess.net
 * epel: mirror.prgmr.com
 * extras: mirror.keystealth.org
 * rpmforge: mirror.us.leaseweb.net
 * rpmfusion-free-updates: mirrors.lug.mtu.edu
 * rpmfusion-nonfree-updates: mirrors.lug.mtu.edu
 * updates: mirror.cisp.com
--> Finished Dependency Resolution
Error: Package: libiio-0.15.g6ecff5d-1.x86_64 (/libiio-0.15.g6ecff5d-Linux-trusty)
           Requires: libiio.so.0()(64bit)
 You could try using --skip-broken to work around the problem
 You could try running: rpm -Va --nofiles --nodigest
[wrthissell@localhost x86_64]$ sudo yum install libiio-0.15.g6ecff5d-Linux-precise.rpm
Loaded plugins: fastestmirror, langpacks, nvidia
Examining libiio-0.15.g6ecff5d-Linux-precise.rpm: libiio-0.15.g6ecff5d-1.x86_64
Marking libiio-0.15.g6ecff5d-Linux-precise.rpm to be installed
Resolving Dependencies
--> Running transaction check
---> Package libiio.x86_64 0:0.15.g6ecff5d-1 will be installed
--> Processing Dependency: libiio.so.0()(64bit) for package: libiio-0.15.g6ecff5d-1.x86_64
Loading mirror speeds from cached hostfile
 * base: mirrors.xtom.com
 * centosplus: mirror.linuxfix.com
 * epel: mirror.prgmr.com
 * extras: mirror.keystealth.org
 * rpmforge: mirror.us.leaseweb.net
 * rpmfusion-free-updates: mirrors.lug.mtu.edu
 * rpmfusion-nonfree-updates: mirrors.lug.mtu.edu
 * updates: repo1.dal.innoscale.net
--> Finished Dependency Resolution
Error: Package: libiio-0.15.g6ecff5d-1.x86_64 (/libiio-0.15.g6ecff5d-Linux-precise)
           Requires: libiio.so.0()(64bit)
 You could try using --skip-broken to work around the problem
 You could try running: rpm -Va --nofiles --nodigest
[wrthissell@localhost x86_64]$ 
same issue with the trusty rpm.

Are you able to post src.rpms so that one may fix the issue with the spec file?

commodo, 2018-09-19:
Hey,

We don't have specifically written spec files for RPMs.
We use cmake to generate them (https://github.com/analogdevicesinc/libiio/blob/master/cmake/LinuxPackaging.cmake ) 

Typically we use Debian based distros to test [specially auto-testing with Travis CI], but we will have to try to integrate some CentOS auto-testing for the RPMs.

We'll take a look at this ; but in the meantime, you could build libiio from source, from the Github repo.

Thanks
Alex

commodo, 2018-09-27:
I should add, that you can find the temp `spec` file in `<build>/_CPack_Packages/Linux/RPM/SPECS`.

rgetz, 2018-10-27:
We are currently building on Centos 6 and 7 - when the deployments get pushed, I will modify the build scripts, so we don't deploy the rpms from Ubuntu, and use the Centos versions instead,

rgetz, 2018-10-27:
@wrthissell : Until things get merged to master, can you try out:

http://swdownloads.analog.com/cse/travis_builds/test_travis_libiio-0.15.gfb82df7-Linux-centos6.rpm
or
http://swdownloads.analog.com/cse/travis_builds/test_travis_libiio-0.15.gfb82df7-Linux-centos7.rpm

Those should work better (I think), since they are built on the specific versions of centos.