analogdevicesinc/libiio #199: Ubuntu 12.04 LTS (Precise Pangolin) reached its regular End of Life on April 28, 2017.

rgetz, 2018-09-30:
should we just remove it from travis-ci (and it's mostly failing), and add something more modern?

Today Travis-ci supports:
Ubuntu 14.04.5 LTS (Trusty Tahr) : EOL April 2019, which we build now, but April is coming soon.
Ubuntu 12.04.5 LTS (Precise Pangolin) : reached its regular End of Life on April 28, 2017. 

and supports these from a docker environment:
Ubuntu 18.04.1 LTS (Bionic Beaver) : EOL April 2023. 
Ubuntu 16.04.5 LTS (Xenial Xerus) : EOL April 2021

Adding this issue, so we start looking at the docker build environment, and eventually get 16.x & 18.x building.

?

commodo, 2018-10-01:
libiio already builds for Ubuntu 16.04 
See https://travis-ci.org/analogdevicesinc/libiio/jobs/435412264

For bionic, we would need to add a docker build, since Travis-CI doesn't have bionic in their inventory yet.

Currently, it's in-progress to add CentOS 6 & 7, since there seem to be some questions & interest about building libiio on CentOS. See PR #196

commodo, 2018-10-01:
Regarding 12.04, I think we could leave it until Travis CI drops it. They do show a banner about it being in the process of being decommissioned, but that looks like it's been there for a few months.

The builds on 12.04 haven't been failing in the last 22 days or so (https://travis-ci.org/analogdevicesinc/libiio/builds ) . They are failing intermittently, but I don't know if we can control that.
Ubuntu 12.04 seems to be one of those old distros that seems to have stuck around for a long time; I guess the best moment to discontinue builds for it, is when Travis-CI no longer supports it.

AlexandraTrifan, 2018-10-01:
I'll start looking into a docker build for Bionic, since Xenial is already up and running.

rgetz, 2018-10-01:
Thanks - I didn't see that Xenial was already building, since it wasn't in the releases (I never checked in the README), but that is since we haven't done a release with it yet.

As for the intermittent failures - dropping old things is just part of maintenance. We shouldn't enable bad practices (using not supported OSes), that the vendor isn't providing security updates for. It has been a year plus already...

-Robin

rgetz, 2018-11-06:
I also noticed that we are seeing:
```
Running builds with Xcode 6.4 in Travis CI is deprecated and will be removed in January 2019. 
If Xcode 6.4 is critical to your builds, please contact our support team at 
support@travis-ci.com to discuss options.
```

https://docs.travis-ci.com/user/reference/osx/#os-x-version

https://www.mathworks.com/support/sysreq.html
only goes back to 10.12; so I don't think it is a big deal to drop it.

at the same time we prune precise, we should update these as well.

commodo, 2018-11-06:
I'm fine with that.
Xcode 6.4 builds take way to long to finish

rgetz, 2018-11-12:
Closing - since this has been covered in some recent pull requests.

commit 6085b3a5 (2018-11-07): .travis.yml: Drop the Travis CI builds for Ubuntu 12.04 LTS
(Precise Pangolin).

Fixes #199.
commit f8d2eb83 (2018-11-07): .travis.yml: Drop the Travis CI builds for Ubuntu 12.04 LTS
(Precise Pangolin).

Fixes #199.
closed 2018-11-12 by rgetz