analogdevicesinc/libiio #200: deb install failures

tfcollins, 2018-10-05:
I believe debs built on travis-ci are not setting some required packaging information. Required fields: https://www.debian.org/doc/debian-policy/ch-controlfields.html#s-binarycontrolfiles

This is resulting in install failures: https://travis-ci.org/analogdevicesinc/libad9361-iio/jobs/437402773#L634

commodo, 2018-10-12:
It's weird that it only happens on Trusty builds.
Xenial & Precise don't seem affected.

tfcollins, 2018-10-12:
We could do a  --force-architecture for now in libad9361 or just set the CPACK_DEBIAN_PACKAGE_ARCHITECTURE in the current cmake build to force that. It shouldn't affect the other builds.

commodo, 2018-10-14:
That still leaves iio-oscilloscope & scopy open for the issue.
I'll think about it a bit.
The problem seems to be travis-ci's fault, not so much ours.

I'll try to take a look.

rgetz, 2018-10-26:
I believe that is the issue (old cmake doesn't set this properly), and we need something like the below in libiio/cmake/LinuxPackaging.cmake

    # Derive architecture
    IF(NOT CPACK_DEBIAN_PACKAGE_ARCHITECTURE)
      FIND_PROGRAM(DPKG_CMD dpkg)
      IF(NOT DPKG_CMD)
        MESSAGE(STATUS "Can not find dpkg in your path, default to i386.")
        SET(CPACK_DEBIAN_PACKAGE_ARCHITECTURE i386)
      ENDIF(NOT DPKG_CMD)
      EXECUTE_PROCESS(COMMAND "${DPKG_CMD}" --print-architecture
        OUTPUT_VARIABLE CPACK_DEBIAN_PACKAGE_ARCHITECTURE
        OUTPUT_STRIP_TRAILING_WHITESPACE
        )
    ENDIF(NOT CPACK_DEBIAN_PACKAGE_ARCHITECTURE)

rgetz, 2018-10-30:
@tfcollins can you try out: http://swdownloads.analog.com/cse/travis_builds/rgetz-patch-2_libiio-0.15.g2071150-Linux-trusty.deb

I think that should have things working.