analogdevicesinc/libiio #204: regex issues in libiio?

rgetz, 2018-10-24:
There are a few channels that have *_available, but no *, which is weird.

For example, now that things are sorting, its easier to tell..

This is OK
dev 'ad9361-phy', channel 'voltage0' (output), attr 'rf_port_select', value 'A'
dev 'ad9361-phy', channel 'voltage0' (output), attr 'rf_port_select_available', value 'A B'

This is not.
dev 'ad9361-phy', channel 'voltage2' (output), attr 'rf_bandwidth_available',
dev 'ad9361-phy', channel 'voltage2' (output), attr 'rf_port_select_available',

dev 'ad9361-phy', channel 'voltage2' (input), attr 'rf_dc_offset_tracking_en',
dev 'ad9361-phy', channel 'voltage2' (input), attr 'rf_port_select_available', 

root@analog:/sys/bus/iio/devices/iio:device1# find ./ -name "*rf_port*" | sort
./in_voltage0_rf_port_select
./in_voltage1_rf_port_select
./in_voltage_rf_port_select_available
./out_voltage0_rf_port_select
./out_voltage1_rf_port_select
./out_voltage_rf_port_select_available

it's either a libiio issue - it should not be adding it when the voltage2_rf_port_select doesn't exist, or a kenel one, where it is missing to populate some attributes.

same thing with:

dev 'ad9361-phy', channel 'voltage2' (input), attr 'filter_fir_en',
dev 'ad9361-phy', channel 'voltage2' (input), attr 'gain_control_mode_available',

and a few others.

-Robin

mhennerich, 2018-10-24:
It's a driver issue, we are aware of, but has been hesitant to fix, in order to not break none libiio based user space apps. 
The issue is following. The original driver had in_voltage0 for RX1 and in_voltage1 for RX2 so IIO convention was to skip the index for things that apply for all in_voltageX channels. 
and use in_voltage_rf_port_select_available for example. Later on we added the AUX ADC and DAC channels getting in_volatge2 for example. So we should have fixed things but we didn't, also the IIO core was not flexible enough for the IIO enum infrastructure we used. In the meantime we could, but thought we break API only once when we mainline this driver. Which hasn't happened so far.

rgetz, 2018-10-24:
Ok - thanks for description. Will close here. 

- Robin

closed 2018-10-24 by rgetz