analogdevicesinc/libiio #218: adc scale attributes include extra newline

rgetz, 2018-11-05:
I don't know if this is a bug in libiio, or in the kernel.

in some kernel drivers, the attributes (accidently?) puts a newline at the end of the field, which gets added, and sent to the entire system...

```
dev 'ad7291', channel 'voltage0' (input), attr 'raw', value '3330'
dev 'ad7291', channel 'voltage0' (input), attr 'scale', value '0.610351562
'
dev 'ad7291', channel 'voltage1' (input), attr 'raw', value '3364'
dev 'ad7291', channel 'voltage1' (input), attr 'scale', value '0.610351562
'
dev 'ad7291', channel 'voltage2' (input), attr 'raw', value '3367'
dev 'ad7291', channel 'voltage2' (input), attr 'scale', value '0.610351562
'
```
rather than

```
dev 'ad7291', channel 'voltage0' (input), attr 'raw', value '3378'
dev 'ad7291', channel 'voltage0' (input), attr 'scale', value '0.610351562'
dev 'ad7291', channel 'voltage1' (input), attr 'raw', value '3364'
dev 'ad7291', channel 'voltage1' (input), attr 'scale', value '0.610351562'
dev 'ad7291', channel 'voltage2' (input), attr 'raw', value '3368'
dev 'ad7291', channel 'voltage2' (input), attr 'scale', value '0.610351562'
dev 'ad7291', channel 'voltage3' (input), attr 'raw', value '1307'
dev 'ad7291', channel 'voltage3' (input), attr 'scale', value '0.610351562'
dev 'ad7291', channel 'voltage4' (input), attr 'raw', value '2510'
dev 'ad7291', channel 'voltage4' (input), attr 'scale', value '0.610351562'
dev 'ad7291', channel 'voltage5' (input), attr 'raw', value '1303'
dev 'ad7291', channel 'voltage5' (input), attr 'scale', value '0.610351562'
dev 'ad7291', channel 'voltage6' (input), attr 'raw', value '1'
dev 'ad7291', channel 'voltage6' (input), attr 'scale', value '0.610351562'
dev 'ad7291', channel 'voltage7' (input), attr 'raw', value '1'
dev 'ad7291', channel 'voltage7' (input), attr 'scale', value '0.610351562'
```

if we want to eat newlines at the end of the line, this works, but I wasn't sure how deep to put things (or change it in the kernel driver).

```
root@analog:/usr/local/src/libiio# git diff
diff --git a/tests/iio_attr.c b/tests/iio_attr.c
index e4d05d7..a994035 100644
--- a/tests/iio_attr.c
+++ b/tests/iio_attr.c
@@ -269,6 +269,10 @@ static void dump_channel_attributes(const struct iio_device *dev,
                        type_name = "input";
 
                ret = iio_channel_attr_read(ch, attr, buf, sizeof(buf));
+               /* strip trailing newlines */
+               if (buf[strlen(buf) - 1] == '\n')
+                       buf[strlen(buf) - 1] = 0;
+
                if (!quiet)
                        printf("dev '%s', channel '%s' (%s), ",
                                        iio_device_get_name(dev),
```

I see it in the xadc and ad7291 scale attributes. If this is a fix-in-the-kernel issue - feel free to close.

-Robin

mhennerich, 2018-11-06:
The driver code that generates the scale is here:
https://github.com/analogdevicesinc/linux/blob/master/drivers/iio/adc/ad7291.c#L377

It let's the IIO core generate the string - so there is nothing special with it.

Also looking at the returned strings - I don't see any extra line breaks!
 
```
root@analog:/sys/bus/iio/devices/iio:device2# hexdump -c name                                                                                                                                               
0000000   a   d   7   2   9   1  \n                                    
0000007
root@analog:/sys/bus/iio/devices/iio:device2# hexdump -c in_voltage_scale                                                                                                                                   
0000000   0   .   6   1   0   3   5   1   5   6   2  \n                
000000c
root@analog:/sys/bus/iio/devices/iio:device2# hexdump -c in_voltage0_raw                                                                                                                                  
0000000   2   3   5   7  \n                                            
0000005
```

There is always a new line:
https://github.com/analogdevicesinc/linux/blob/master/drivers/iio/industrialio-core.c#L664

So I really don't know where those extras are coming from.

pcercuei, 2018-11-06:
This is supposed to eliminate the \n: https://github.com/analogdevicesinc/libiio/blob/master/local.c#L719
I'd suggest to add some debug there, maybe the data ends with \n\r and only the \r is removed.

rgetz, 2018-11-10:
I will have a look. @pcercuei - thanks for the pointer

rgetz, 2018-11-12:
when I look at things, I see two newlines coming from the kernel.

```
analog@analog:~$ hexdump -c /sys/bus/iio/devices/iio:device0/name
0000000   a   d   7   2   9   1  \n                                    
0000007
analog@analog:~$ hexdump /sys/bus/iio/devices/iio:device0/in_voltage_scale -c
0000000   0   .   6   1   0   3   5   1   5   6   2  \n  \n            
000000d
analog@analog:~$ uname -a
Linux analog 4.9.0-g0fa1a63 #18 SMP PREEMPT Fri Jan 26 08:34:30 GMT 2018 armv7l armv7l armv7l GNU/Linux
```

this fixes things:
```
diff --git a/local.c b/local.c
index d05afb6..febd9ec 100644
--- a/local.c
+++ b/local.c
@@ -715,8 +715,13 @@ static ssize_t local_read_dev_attr(const struct iio_device 
                return -errno;
 
        ret = fread(dst, 1, len, f);
-       if (ret > 0)
+
+       if (ret > 0) {
                dst[ret - 1] = '\0';
+               while (dst[strlen(dst) - 1] == '\n' ||
+                               dst[strlen(dst) - 1] == '\r' )
+                       dst[strlen(dst) - 1] = '\0';
+       }
        fflush(f);
        if (ferror(f))
                ret = -errno;
```

I can make a branch/patch if no one objects,

pcercuei, 2018-11-12:
The kernel should never give you a \r so I think there's no need for that.

For the patch: I think you can do it without calls to strlen().

rgetz, 2018-11-12:
Closing this, and following up in pull request.

closed 2018-11-12 by rgetz