analogdevicesinc/libiio #219: iio_channel_get_modifier and iio_channel_get_type seems to work incorrectly.

mikexing2018, 2018-11-07:
In my driver, I have 6 channels.

3 of the 6 channels have IIO_ANGL_VEL as type and IIO_MOD_X, IIO_MOD_Y, IIO_MOD_Z as channel_2 (modifier) respectively.

The other 3 channels are IIO_ACCEL and have IIO_MOD_X, IIO_MOD_Y, IIO_MOD_Z as channel_2 (modifier) respectively.

Using libiio's iio_channel_get_modifier and iio_channel_get_type, I am not getting the correct results:

For the IIO_ACCEL channels, I am getting 3 as type (which is correct), and 0 for all the modifiers (incorrect).
For the IIO_ANGL_VEL channels, I am getting a 12 for channel type (incorrect), and 0 for all the modifiers (incorrect)

commodo, 2018-11-07:
Which device are you using this with ?
We could try to reproduce it locally and see what happens.

mikexing2018, 2018-11-07:
This is an IMU device. For confidentiality purpose, I cannot disclose the exact device.

I think this issue should not be device specific. Maybe there is some general problem with those APIs for getting the channel type and modifier? Has anyone ever tested?

commodo, 2018-11-07:
What version of libiio are you using ?

mikexing2018, 2018-11-07:
@pcercuei Yes. I am using local backend. Here are the sequence that I am doing:
iio_create_local_context()
iio_context_find_device
iio_device_get_channels_count
iio_device_get_channel
iio_device_get_type
iio_device_get_modifier

mikexing2018, 2018-11-07:
@commodo 

https://github.com/analogdevicesinc/libiio/archive/v0.15.tar.gz

pcercuei, 2018-11-07:
Maybe a filename parsing error. What are the IDs of the channels, as reported by iio_info?

mikexing2018, 2018-11-07:
The iio_channel_get_id are returning the following values for the 6 channels:
accel_x, accel_y, accel_z, gyro_x, gyro_y, gyro_z

seems correct.

mikexing2018, 2018-11-07:
The issue is in iio_channel_init_finalize(). There is string parsing errors there for both channel type and channel modifier. For modifier, I don't know why there is an assumption that the modifier needs to be followed by an underscore. For channel type. it's suppose to break when it finds the right type rather than keep looping.

pcercuei, 2018-11-07:
For the modifier, you're right. The check for the underscore is bogus.

For the type, it should break, but that's not the cause of the problem. The condition for the "continue" right above where chn->type is set looks really wrong; the second && should be a || instead (with proper parenthesis set around it).

mhennerich, 2018-11-07:
@mikexing2018 you have a test case - do you mind sending a Pull Request?

mhennerich, 2018-11-08:
Request for test: 
Can you test the fix in the https://github.com/analogdevicesinc/libiio/tree/rft-issue-219 branch?

mikexing2018, 2018-11-09:
Sure

mhennerich, 2018-11-12:
> Sure

Does it fix it?

commodo, 2018-11-19:
@mikexing2018 
ping

mikexing2018, 2018-11-19:
I don't have access to the testing environment this week. Will need to delay until the week of Dec 26th.

commodo, 2018-11-19:
ok :)
thanks for the feedback;
we can wait

mhennerich, 2018-11-19:
Dec 26th? I don't think we can wait that long. 
The PACKRF has a ADIS16460 type IMU, we should be able to quickly test it there. 
Alex can you take a look?

commodo, 2018-11-20:
Yep
Will take a look

Just need to take a look into some build issues first.

mhennerich, 2018-11-21:
Ok - i did some test using a ADIS16460.
		
```
root@analog:/sys/bus/iio/devices/iio:device4# ls -al
total 0
drwxr-xr-x 6 root root    0 Jan  1  1970 .
drwxr-xr-x 6 root root    0 Jan  1  1970 ..
drwxrwxrwx 2 root root    0 Jan  1  1970 buffer
-rw-rw-rw- 1 root root 4096 Jan  1  1970 dev
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_accel_scale
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_accel_x_raw
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_accel_y_raw
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_accel_z_raw
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_anglvel_scale
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_anglvel_x_raw
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_anglvel_y_raw
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_anglvel_z_raw
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_temp0_offset
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_temp0_raw
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_temp0_scale
-rw-rw-rw- 1 root root 4096 Jan  1  1970 name
lrwxrwxrwx 1 root root    0 Jan  1  1970 of_node -> ../../../../../../../../firmware/devicetree/base/amba/spi@e0007000/adis16460@0
drwxrwxrwx 2 root root    0 Jan  1  1970 power
-rw-rw-rw- 1 root root 4096 Jan  1  1970 sampling_frequency
drwxrwxrwx 2 root root    0 Jan  1  1970 scan_elements
lrwxrwxrwx 1 root root    0 Jan  1  1970 subsystem -> ../../../../../../../../bus/iio
drwxrwxrwx 2 root root    0 Jan  1  1970 trigger
-rw-rw-rw- 1 root root 4096 Jan  1  1970 uevent
root@analog:/sys/bus/iio/devices/iio:device4# ls -al scan_elements/
total 0
drwxrwxrwx 2 root root    0 Jan  1  1970 .
drwxr-xr-x 6 root root    0 Nov 21 08:12 ..
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_accel_x_en
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_accel_x_index
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_accel_x_type
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_accel_y_en
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_accel_y_index
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_accel_y_type
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_accel_z_en
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_accel_z_index
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_accel_z_type
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_anglvel_x_en
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_anglvel_x_index
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_anglvel_x_type
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_anglvel_y_en
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_anglvel_y_index
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_anglvel_y_type
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_anglvel_z_en
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_anglvel_z_index
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_anglvel_z_type
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_temp0_en
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_temp0_index
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_temp0_type
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_timestamp_en
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_timestamp_index
-rw-rw-rw- 1 root root 4096 Jan  1  1970 in_timestamp_type

root@analog:/sys/bus/iio/devices/iio:device4/scan_elements# grep "" *
in_accel_x_en:0
in_accel_x_index:3
in_accel_x_type:be:s32/32>>0
in_accel_y_en:0
in_accel_y_index:4
in_accel_y_type:be:s32/32>>0
in_accel_z_en:0
in_accel_z_index:5
in_accel_z_type:be:s32/32>>0
in_anglvel_x_en:0
in_anglvel_x_index:0
in_anglvel_x_type:be:s32/32>>0
[... 120 lines cut ...]
			accel_z:  (input, index: 5, format: be:S32/32>>0)
				type=3, modifier=3
			2 channel-specific attributes found:
				attr  0: raw value: 255459327
				attr  1: scale value: 0.000000037
			temp0:  (input, index: 6, format: be:S16/16>>0)
				type=9, modifier=0
			3 channel-specific attributes found:
				attr  0: offset value: 500
				attr  1: scale value: 50
				attr  2: raw value: 227
			timestamp:  (input, index: 7, format: le:S64/64>>0)
				type=2147483647, modifier=0
		1 device-specific attributes found:
				attr  0: sampling_frequency value: 2048.000000
		1 buffer-specific attributes found:
				attr  0: watermark value: 1
		4 debug attributes found:
				debug attr  0: flash_count value: 157
				debug attr  1: product_id value: 16460
				debug attr  2: serial_number value: 0x09da
				debug attr  3: direct_reg_access value: 0x9D
		Current trigger: trigger0(adis16460-dev4)		
```
-Michael

commit 25c20d67 (2018-11-21): channel: Fix #219 get_modifier and get_type seems to work incorrectly

Untested attempt to fix issue #219:
iio_channel_get_modifier and
iio_channel_get_type seems to work incorrectly.
commit effcce83 (2018-11-21): Merge pull request #224 from analogdevicesinc/rft-issue-219

channel: Fix #219 get_modifier and get_type seems to work incorrectly
commit e868971c (2018-11-21): Bump to version v0.16

Changelog:
20c9079 CI/travis/inside_bionic_docker.sh: re-use the make_linux logic
cd1be7d build,travis: parametrize library name
2b4503d README.md : Update with links to doc
cf87e0a travis-ci: install graphviz
2f3632f doxygen: check if building on a case senstive file system.
a18ff6e doxygen: If dot exists, use it.
cb28721 appveyor: install graphviz and fix doxygen version
550f4b8 README: Update with latest build changes
540c96b deployment : export the LDIST var which is needed in other scripts
de8356b fix snprintf warnings from gcc 8
24c7f00 .travis.yml: Add new Xcode v [... cut]
commit 297b1e25 (2018-11-21): channel: Fix #219 get_modifier and get_type seems to work incorrectly

Untested attempt to fix issue #219:
iio_channel_get_modifier and
iio_channel_get_type seems to work incorrectly.
closed 2018-11-21 by mhennerich