analogdevicesinc/libiio #222: couldnt connect to libiio server

dhruvmar, 2018-11-11:
I am trying to run a simple lte example on zedboard from avnet with ad9361 FMCOMMs2 but always get the following message. couldnt connect to libiiio server. iiod server is running on zedboard. 
I ran the git clone " ...../libiio.git" on zedboard ... it gives me a fatal error of emptyreply from server.

I can git colen it on a normal ubuntu or fedora machin but on zedboard its not working.

rgetz, 2018-11-12:
it could be possible (likely) that the date is set wrong. (if so, the ssh keys fail).

run:
```
sudo /usr/bin/htpdate -ts www.analog.com
sudo adi_update dev
```

some early versions of the SD card distribution used analog.com to set the time - but it doesn't work anymore(because of some DoS attacks, regular http headers don't come back anymore on analog.com; but do on www.analog.com).

rgetz, 2018-11-15:
Support for libiio found on the ADI Zynq/Intel SoC images is normally done on:
https://ez.analog.com/linux-device-drivers/linux-software-drivers

If you have further questions - please ask there.

closed 2018-11-15 by rgetz