analogdevicesinc/libiio #236: IIO event handling support

mhennerich, 2018-11-22:
libiio should support all aspects of IIO devices. Right now events are kind of unsupported.

Support of events was asked here:
https://ez.analog.com/linux-device-drivers/linux-software-drivers/f/q-a/104138/does-libiio-support-iio-events

This issue/task is to track this development. If someone want's to work on this feel free to contribute.

jeff-dagenais, 2019-04-26:
Hi Micheal,

Any work by anyone on this? What are people doing right now if one needs to configure thresholds in a DAC for example and wait for the event?

mhennerich, 2019-04-26:
Sorry to say - Nothing done so far.

jeff-dagenais, 2019-04-26:
And about my second question? Directly access the pseudo-files I suppose (e.g. `/sys/devices/platform/amba/ff020000.i2c/i2c-0/0-0048/iio:device3/events/in_voltage0_thresh_*`) ?

vdayananz, 2023-05-10:
@mhennerich , 

It would be nice to get support for iio-events.  When do you think there would be APIs that can configure and monitor events ?

pcercuei, 2023-10-11:
@vdayananz still around?
I think I will have something to test before the end of the month.

edrose-tewke, 2023-11-03:
I'm interested in events support if it's been implemented, however my ability to do testing is going to be limited by my lack of access to hardware.

pcercuei, 2023-11-03:
@edrose-tewke I'm working on it right now, it should be ready next week!