analogdevicesinc/libiio #257: iio_readdev gives corrupted data on native windows (7 and 10)

meds7, 2019-01-28:
Hi, 
I tested iio_readdev data retrieval from a network backend which is able to send data at 125Mbit/s as iio device. It appears that both W7 and W10 native libiio (0.17 tested) results in corrupted data retrieval. 
Tests on the same W10 machine but using WSL works great.

Henk Medenblik

commodo, 2019-02-04:
could you give us some input on the libiio you used ?
did you build it yourself, or did you use an installer built by us ?

thanks
Alex

meds7, 2019-02-04:
Dear Alexandru,

For this test I used the installer which can be found at https://ci.appveyor.com/api/projects/analogdevicesinc/libiio/artifacts/libiio-setup.exe?branch=master

I hope that helps to identify it. Let me know if I can help you further to identify the problem.

With kind regards !
Henk

rgetz, 2020-02-04:
OK - we will try to have a look at this before the next release. @dNechita - do you have time?

dNechita, 2020-02-05:
> OK - we will try to have a look at this before the next release. @dNechita - do you have time?

Yes, I will look into this before we do the next release.

dNechita, 2020-02-06:
Hi @meds7 

Sorry for this very late reply!

Do you remember which bash did you use to run the iio_readdev example on Windows7/10?

rgetz, 2020-02-07:
I think he said “W7 and W10 native libiio“. So no bash - just a native windows prompt. (cmd)

meds7, 2020-02-07:
Hi @dNechita , @rgetz 

Indeed, the issue is with native Windows as @rgetz  pointed out. Unfortunately my piece of hardware has been shipped to the end customer already for a long time ago so I can’t verify any new changes.

Regards,
Henk

dNechita, 2020-02-07:
The issue in not in the libiio library but in the iio_readdev program.
I have the fix in the PR. It was an issue of how stdio works on windows.

closed 2020-02-07 by dNechita

rgetz, 2020-02-07:
thanks for fixing.

commit 4aa00554 (2020-02-07): iio_readdev: Fix corrupt data that was being captured on Windows

The captured data is sent to stdout but during this process (on Windows)
any byte that has the same value as line feed character (LF) gets
converted to CF-LF characters. This dissrupts the alignment of bytes
that represent the samples of the device.

Fixes #257.
commit bb609749 (2020-02-07): iio_readdev: Fix corrupt data that was being captured on Windows

The captured data is sent to stdout but during this process (on Windows)
any byte that has the same value as line feed character (LF) gets
converted to CF-LF characters. This dissrupts the alignment of bytes
that represent the samples of the device.

Fixes #257.
commit f7c5e7f6 (2020-02-07): iio_readdev: Fix corrupt data that was being captured on Windows

The captured data is sent to stdout but during this process (on Windows)
any byte that has the same value as line feed character (LF) gets
converted to CF-LF characters. This dissrupts the alignment of bytes
that represent the samples of the device.

Fixes #257.