analogdevicesinc/libiio #263: Use strerror() wherever possible

commodo, 2019-03-07:
Some error reports are "I get code X", which prompts questions.
We could mitigate this by putting `strerror()` wherever possible, to also print strerror() strings.
Especially in examples

rgetz, 2019-07-19:
I think iio_attr and iio_info already do this.

Which other examples did you have in mind?

pcercuei, 2019-07-20:
There's ```iio_strerror``` btw.

rgetz, 2019-07-23:
Ok - a little check...

```
rgetz@brain:~/github/libiio$ grep iio_strerr ./tests/* -c
./tests/CMakeLists.txt:0
./tests/iio_adi_xflow_check.c:1
./tests/iio_attr.c:10
./tests/iio_genxml.c:0
./tests/iio_info.c:6
./tests/iio_readdev.c:3
./tests/iio_reg.c:0
./tests/iio_writedev.c:3
```

which isn't too bad - 

but

```
rgetz@brain:~/github/libiio$ grep iio_strerr ./examples/* -c
./examples/ad9361-iiostream.c:0
./examples/ad9371-iiostream.c:0
./examples/adrv9009-iiostream.c:0
./examples/dummy-iiostream.c:0
./examples/iio-monitor.c:0
./examples/Makefile:0
```

those need some work/care...

commodo, 2019-07-24:
As I remember, I created this issue because of some repeated reports on EZ would be "running example/code X, I get -22"; 

I forget which examples they were providing, but likely some of them were `ad9361-iiostream.c` [for Pluto and other boards with AD9361] and  `ad9371-iiostream.c` .
Naturally, some of our replies were: "-22 is Invalid Argument" or "-110 is Timedout ", etc;  so, we tell people to check things, but usually after we look for the `strerror()` value of that error code.

Not sure how much reduction in EZ questions [of this sort] we'd get by doing this; I just added it here, so I won't forget about it.

commodo, 2020-04-02:
i guess i'll close this for now;
it's a bit too generic;
and probably won't solve enough EZ issues anyway;

closed 2020-04-02 by commodo