analogdevicesinc/libiio #264: buildroot LTS now includes broken v0.14 libiio

skrap, 2019-03-08:
I noticed that buildroot upgraded its libiio distribution to 0.14, which for me broke my IIO buffer-based devices.  I haven't tracked down what went wrong exactly (writing to the `/enable` sysfs file returns errno 2, inside of iio_buffer_create) but 0.10 (the previous version in buildroot) and 0.17 (current libiio) both work perfectly.

I'm no expert on libiio, so I don't feel confident speaking up to the buildroot folks, but it seems like if v0.14 is broken in some critical way, that someone might want to recommend that buildroot not ship it in the LTS release.  The LTS release is only a few days old, so it's a good opportunity to fix this (if it is indeed an issue) before too many bad copies of libiio get out into the world.

mhennerich, 2019-03-10:
Sounds like this issue: https://github.com/analogdevicesinc/libiio/commit/08929722633d83f710b4620a1cd65ef15b076446

which got fixed in v0.15. We'll send a patch to update to v0.15.

-Michael

commodo, 2019-03-13:
Typically in releases, maintainers prefer patches vs version updates.
But, I'll see if they will accept a version update.

commodo, 2019-03-19:
hey,

i just got replies from buildroot that they've update libiio to version 0.15 in both master and LTS (2019.02.x)
please test and see if it fixes this for you;

i'll close this issue in order to unload some of my stuff that i track;
but if there are issues feel free to re-open

thanks
alex

closed 2019-03-19 by commodo

skrap, 2019-03-19:
Many thanks!