analogdevicesinc/libiio #265: Syntax for IIOD open command using Nc

bartona, 2019-03-10:
I am trying to communicate with an iiod server simply over a terminal using commands over netcat.  I am able to use the commands "PRINT", "HELP", "VERSION", "SET iio:device0 BUFFERS_COUNT 4", and "TIMEOUT 100" with no issue.  However the command "OPEN iio:device0 128 00000001" fails with error code -22.  Is this the correct syntax for the OPEN command?  There isn't an example listed in the libiio wiki.

pcercuei, 2019-03-10:
I looks like the correct command. -22 means "invalid value", though. Does it work if you create a buffer using the network backend of libiio?

rgetz, 2020-01-14:
Closing since we really don't support iiod manual commands - if @bartona still needs some help - feel free to re-open things.

closed 2020-01-14 by rgetz