analogdevicesinc/libiio #266: iiod network backend: duplicating the stream and server side demultiplexing

gwendalcr, 2019-03-13:
Today, the network backed of IIOD is duplicating the device data stream to the connected clients using the same device at the same time. 
It would be nice if the server would [demultiplex ](https://wiki.analog.com/resources/tools-software/linux-software/libiio_internals#server-side_and_client-side_de-multiplexing), based on the clients need.
The server would return the running average of continuous sensors.
Client would define the minimum frequency and the server would send samples at up to twice that frequency.

For instance, if a sensor supports data rate of 13Hz, 26Hz, 52Hz, 104Hz.
If client A asks for 100Hz, clients B asks 40Hz, iiod would sends a stream of samples at 104Hz for A, 52Hz for B, each samples to B would be the average of two samples. 
When client A disconnect, iiod would change the data rate to 52Hz and sends every sample to B.

Has this enhancement been considered?

rgetz, 2019-07-23:
I don't think so - since most of the use cases are really meant for single clients.

Right now, if two clients attach, one gets buffer A, the other gets buffer B.

There isn't any effort to make sure that both get continuously streamed samples. It would put a large processing requirement on the platform running iiod.

pcercuei, 2019-07-23:
Actually it is already implemented and working, although not as specified above.

If two clients attach and try to read from the same device, IIOD will send its iio_buffer to both. By default the buffer will contain the samples of all the channels enabled within one or the other client and the clients are responsible for demultiplexing the data. With the -D option, iiod will demultiplex the data itself and only send to a client the samples that correspond to its enabled channels.

rgetz, 2020-01-13:
If there is still an issue - please let us know.

closed 2020-01-13 by rgetz