analogdevicesinc/libiio #268: Device attributes returning multi-line strings can cause error -EIO

mhennerich, 2019-04-23:
Device attributes returning multi-line strings can cause error -EIO (-5) when read via IIOD.

dev 'ad9361-phy', attr 'gain_table_config', value :ERROR: Input/output error (-5) 

This issue was reported here:
https://ez.analog.com/linux-device-drivers/linux-software-drivers/f/q-a/106570/gain-table-config

rgetz, 2019-07-19:
confirmed:
```
 iio_attr -a -d ad9361-phy gain_table_config
Using auto-detected IIO context at URI "usb:4.8.5"
dev 'ad9361-phy', attr 'gain_table_config', value :ERROR: Input/output error (-5)
```

on local, (on pluto) things work:
```
iio_attr -d ad9361-phy gain_table_config
```

rgetz, 2020-01-19:
Ok - so I think @dNechita suggestion in #345 fixes things completely (almost).

The local is actually broken for the same reason (1k buffers)

```
# uname -a
Linux pluto 4.14.0-42540-g387d584 #301 SMP PREEMPT Wed Jul 3 15:06:53 CEST 2019 armv7l GNU/Linux
# iio_attr -a -d ad9361-phy gain_table_config
Using auto-detected IIO context at URI "local:"
dev 'ad9361-phy', attr 'gain_table_config', value :'<gaintable AD9361 type=FULL dest=3 start=1300000000 end=4000000000>
-3, 0x00, 0x00, 0x20
-3, 0x00, 0x00, 0x00
.... <snip> ....
39, 0x44, 0x27, 0x00
40, 0x44, 0x28, 0x00'
# 
```

the gain table has 70+ entries...

If I look at the actual sysfs, I see:
```
# pwd
/sys/bus/iio/devices/iio:device1
# cat gain_table_config | wc -c
4096
```
so, it's at least a 4k buffer that it needs to be put in.

Making the change that Dan suggested (just bigger) fixes things.

Let's see what it really is (local):
```
# cat gain_table_config 
<gaintable AD9361 type=FULL dest=3 start=1300000000 end=4000000000>
-3, 0x00, 0x00, 0x20
-3, 0x00, 0x00, 0x00
-3, 0x00, 0x00, 0x00
-2, 0x00, 0x01, 0x00
..... <snip> ....
69, 0x6D, 0x38, 0x20
70, 0x6E, 0x38, 0x20
71, 0x6F, 0x38, 0x20
</gaintable>
# 
```

```
rgetz@brain:~/github/libiio/build$ ./tests/iio_attr -a -d ad9361-phy gain_table_config
Using auto-detected IIO context at URI "usb:3.10.5"
dev 'ad9361-phy', attr 'gain_table_config', value :'<gaintable AD9361 type=FULL dest=3 start=1300000000 end=4000000000>
-3, 0x00, 0x00, 0x20
-3, 0x00, 0x00, 0x00
-3, 0x00, 0x00, 0x00
.... <snip> ....
70, 0x6E, 0x38, 0x20
71, 0x6F, 0x38, 0x20
</gaintable>
'
```

which looks like it working for me (pull request #348 )

dNechita, 2020-01-20:
About this:
```
cat gain_table_config | wc -c
4096
```

The content of gain_table_config is around 1650 bytes so you should not be getting 4096. I've seen that extra bytes are zeros but could be garbage data.  There's a issue in the ad9361-phy driver, @commodo is looking into that.
On local, things work ok but it confuses the iiod.
I believe that this issue needs to be closed only when the we do a new release of linux.

mhennerich, 2020-01-20:
Gain table uses a sysfs bin attribute, which size is typically set when it's being created.
Since we allow the gaintable to be modified, it's size is set to a PAGE, which is guaranteed to be big enough . The memory is zero allocated, so there won't be any garbage data trailing.

commodo, 2020-01-20:
> The memory is zero allocated, so there won't be any garbage data trailing.

There is some trailing garbage.
Will create a fix for the kernel.

dNechita, 2020-01-20:
I've spoke without testing when I said: "On local, things work ok but it confuses the iiod." Looks like PR #348 is enough to fix this issue.
Regarding the 4096 bytes issue, that is something I've observed while debugging and seemed as an inconsistency (all iio attributes I've worked with return exactly the size of their content).

One conclusion is that libiio behaves differently on local than on remote when client code provides a smaller buffer than the actual content of an iio attribute. On local, the content is truncated to fit the given space in buffer while on remote the ERROR: Input/output error (-5) is returned.

rgetz, 2020-01-20:
yeah, a little shell : 

```
!#/bin/sh

cd /sys/bus/iio/devices
for device in $(ls /sys/bus/iio/devices) ; do
	cd $device
	for attr in $(find ./ -maxdepth 1 -perm -o=r -type f) ; do
		len=$(cat $attr | wc -c)
		echo "$len $device/$attr"
	done
	cd ..
done
```

piped to sort:
```
# ./foo  | sort -n
```

confirms that 4096 is the largest for now (which was fixed by @commodo 's kernel fix.

```
78 iio:device1/./rx_path_rates
78 iio:device1/./tx_path_rates
95 iio:device1/./in_voltage_rf_port_select_available
99 iio:device1/./rssi_gain_step_error
156 iio:device2/./uevent
161 iio:device1/./uevent
182 iio:device4/./uevent
183 iio:device0/./uevent
204 iio:device3/./uevent
4096 iio:device1/./gain_table_config
# 
```

rgetz, 2020-01-28:
Ok - closing this, since I think it’s all fixed now. Let me know if it isn’t. 

We should include something like the above as a kernel driver check.

@dNechita , can you open a new issue that describes the local/network difference?

closed 2020-01-28 by rgetz