analogdevicesinc/libiio #274: Improve examples and add python examples

nullr0ute, 2019-05-13:
It would be useful to add a general example, like a simple weather station, where it reads and displays any temp/pressure/humidity sensor to show how to generally use libiio to access general sensors. A version of the example for both C and the bindings, such as python, to show a good example of getting started.

rgetz, 2020-01-07:
I have been working on #289 which does provide specific code snippets (in C and python) for people who are just getting started. They aren't "complete" examples (like a weather station), but are functioning snippets that can be stitched together to make something more complete.

Comments welcome

-Robin

rgetz, 2020-02-03:
For python examples which use libiio, check out:

https://github.com/analogdevicesinc/pyadi-iio

I’m going to close this, and accept python example pull requests. :)

closed 2020-02-03 by rgetz