analogdevicesinc/libiio #275: Semantic of IIO_CHAN_INFO_SCALE, IIO_CHAN_INFO_OFFSET and IIO_CHAN_INFO_HARDWAREGAIN

matejk, 2019-05-17:
My colleagues and myself are developing a few Linux kernel IIO drivers for a piece of hardware that comprises few ADC, DAC chips, hardware attenuators and Zynq for data processing in FPGA.

The whole board is presented to an user as a fast 8-bit dual channel digital input, where the details of internal settings for gain tables and attenuation are handled by a few Linux kernel modules.

At this point I came across several attributes, most importantly `IIO_CHAN_INFO_SCALE` and `IIO_CHAN_INFO_HARDWAREGAIN`. I studied IIO documentation and several kernel modules, however I was not able to definitely conclude the semantic meaning of these scaling attributes.

My understanding is that `IIO_CHAN_INFO_SCALE` is a multiplier to convert raw (8 bit in our case) value to some physical unit (milli Volts for voltage type channels). Many drivers report constants for this value and multiplication of max raw value with the scale yields some round number like 2V, 20V. What does that value represent? Is this input voltage range of an ADC, for example?

My assumption is that `IIO_CHAN_INFO_HARDWAREGAIN` represents attenuations and gains on the analogue chain before the digitalisation happens. Is that true?

The application that reads raw data via IIO channels can then multiply them with `IIO_CHAN_INFO_SCALE` and `IIO_CHAN_INFO_HARDWAREGAIN` to get the actual input voltage on the input probes. Is my understanding correct?

What is the role of `IIO_CHAN_INFO_OFFSET`?

Thank you very much for your help.

rgetz, 2019-07-19:
Most userspace applications use (error checking removed): 
```
value = read_double_attr(chn, "raw");
value += read_double_attr(chn, "offset");
value *= read_double_attr(chn,  "scale");
```
at the end, value should be: millivolts, or milli-C, or millamps or some other SI unit. 

offset in sysfs is populated by IIO_CHAN_INFO_OFFSET
scale in sysfs is populated by IIO_CHAN_INFO_SCALE

an example:
https://github.com/analogdevicesinc/iio-oscilloscope/blob/master/plugins/dmm.c#L322

matejk, 2019-07-22:
Thank you for the clarification. I still don't know the exact difference between HARDWARE_GAIN and SCALE?

Does my understanding as written above make sense?

mhennerich, 2019-07-22:
https://www.kernel.org/doc/Documentation/ABI/testing/sysfs-bus-iio

In most situations _scale is the preferred attribute. Hardwaregain was introduced with the first VGA/PGA amplifiers in IIO.    

What		/sys/bus/iio/devices/iio:deviceX/out_voltageY_hardwaregain
What:		/sys/bus/iio/devices/iio:deviceX/in_intensity_red_hardwaregain
What:		/sys/bus/iio/devices/iio:deviceX/in_intensity_green_hardwaregain
What:		/sys/bus/iio/devices/iio:deviceX/in_intensity_blue_hardwaregain
What:		/sys/bus/iio/devices/iio:deviceX/in_intensity_clear_hardwaregain
KernelVersion:	2.6.35
Contact:	linux-iio@vger.kernel.org
Description:
		Hardware applied gain factor. If shared across all channels,
		<type>_hardwaregain is used.

What:		/sys/bus/iio/devices/iio:deviceX/in_voltageY_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_voltageY_i_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_voltageY_q_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_voltageY_supply_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_voltage_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_voltage_i_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_voltage_q_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_voltage-voltage_scale
What:		/sys/bus/iio/devices/iio:deviceX/out_voltageY_scale
What:		/sys/bus/iio/devices/iio:deviceX/out_altvoltageY_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_currentY_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_currentY_supply_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_current_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_currentY_i_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_currentY_q_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_current_i_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_current_q_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_accel_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_accel_peak_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_anglvel_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_energy_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_distance_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_magn_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_magn_x_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_magn_y_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_magn_z_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_rot_from_north_magnetic_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_rot_from_north_true_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_rot_from_north_magnetic_tilt_comp_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_rot_from_north_true_tilt_comp_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_pressureY_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_pressure_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_humidityrelative_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_velocity_sqrt(x^2+y^2+z^2)_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_illuminance_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_countY_scale
What:		/sys/bus/iio/devices/iio:deviceX/in_angl_scale
KernelVersion:	2.6.35
Contact:	linux-iio@vger.kernel.org
Description:
		If known for a device, scale to be applied to <type>Y[_name]_raw
		post addition of <type>[Y][_name]_offset in order to obtain the
		measured value in <type> units as specified in
		<type>[Y][_name]_raw documentation.  If shared across all in
		channels then Y and <x|y|z> are not present and the value is
		called <type>[Y][_name]_scale. The peak modifier means this
		value is applied to <type>Y[_name]_peak_raw values.

rgetz, 2019-07-23:
@mhennerich : is it fair to also say that gain is nominally/typically/ in dB, and scale is linear?

? At least that is how it is on the RF VGAs (gain is in dB).

mhennerich, 2019-07-23:
Yes

mhennerich, 2019-07-30:
assumed answered - closed

closed 2019-07-30 by mhennerich

matejk, 2019-07-30:
@mhennerich, @rgetz: Thank you very much.