analogdevicesinc/libiio #277: iiod support for UNIX domain sockets and VSOCK

egranata, 2019-05-23:
The iiod server loop currently creates an `AF_INET` socket to listen for connections.

We would like to extend it to be able to use `AF_UNIX` or `AF_VSOCK` socket objects. This could be driven by configuration options such that the server would start in either one of the inet, unix, vsock modes.

rgetz, 2019-07-19:
Just so everyone is on the same page:

AF_UNIX creates filesystem objects and it only works between processes on the same host - so wouldn't it make more sense to use a local context for iiod if it is on the same machine?

AF_VSOCK is for hypervisor-guest communication; is the iio device driver on the guest OS? and iiod is in the hypervisor? or hypervisor is controlling the hardware (and runs iiod), and the guest OSes connect to it?

What are the use cases?

egranata, 2019-07-30:
> Just so everyone is on the same page:
> 
> AF_UNIX creates filesystem objects and it only works between processes on the same host - so wouldn't it make more sense to use a local context for iiod if it is on the same machine?
> 

Doing this would allow us to limit access to iiod based on file permissions (i.e. we could chown/chgrp the socket file such that not everything running on the system can connect to iiod).

> AF_VSOCK is for hypervisor-guest communication; is the iio device driver on the guest OS? and iiod is in the hypervisor? or hypervisor is controlling the hardware (and runs iiod), and the guest OSes connect to it?

iiod would be running on the host where the sensors live, and the client would be running inside a VM and connecting to iiod to access the host's sensors

> What are the use cases?

For Chrome OS devices, we want to be able to run things inside a VM and have them access the sensors running on the host using the same interface (libiio) as we would if the VM boundary wasn't there

vilasrk, 2022-01-31:
@egranata 
Does vsock connections alive after Device suspend / Resume?

vilasrk, 2022-03-14:
@rgetz 
Any plans to port VSOCK for IIOD?