analogdevicesinc/libiio #280: How to increase number of samples in a buffer while still sampling one sample at a time?

mikexing2018, 2019-05-31:
In my kernel code, I am continuously pushing to the iio buffer every 5ms.

In my user space, I have a process that leverages libiio to read one sample from the buffer at a time.My problem is that when system is under certain spike load, the user space process is not able to consume from the iio buffer fast enough. So I wish to increase my buffer size while still reading one sample at a time. I have tried setting "num_samples" to something like 4. The problem with that is iio_buffer_refill() blocks until it receives the entire 4 samples. This kind of latency is not acceptable in my application.

General outline of my code:

...
buffer = iio_device_create_buffer(device, num_samples, false);  // num_samples = 1
...
while(true) {
   ssize_t nbytes = iio_buffer_refill(ctx->buffer);
   .. Do something with the data I read
}

pcercuei, 2019-05-31:
Try with ```iio_device_set_kernel_buffers_count```. It allows you to change the number of buffers used. The default is 4 IIRC.

mikexing2018, 2019-05-31:
Thank you. I do see this changes the "/.../iio::device1/..../buffer/length" attribute

Can you help me understanding why this function implies that there are multiple buffers? From my limited understanding, there is only one iio buffer and the length parameter specifies how many data samples it can hold.

reference: https://www.kernel.org/doc/html/latest/driver-api/iio/buffers.html
"length, the total number of data samples (capacity) that can be stored by the buffer."

pcercuei, 2019-06-01:
The high-speed interface, made to run with high-speed ADCs and DACs, uses multiple kernel buffers. For the low-speed interface, only one kernel buffer of size ```num_samples * nb_buffers``` is used.

With the default nb_buffers value of 4, and with one-sample buffers, you will start losing samples if you don't dequeue in less than 20ms (4 * 5ms). If you increase nb_buffers to e.g. 16, you would start losing samples after 80ms, so that would help you absorb those system load spikes.

mikexing2018, 2019-06-06:
@pcercuei Thanks for the help. Increasing the number of buffers helped samples lost!

closed 2019-06-06 by mikexing2018