analogdevicesinc/libiio #281: Expected iio buffer latency?

mikexing2018, 2019-06-06:
I have a case of high unexpected latency when using the iio buffer.

My iio buffer has a length of 2. My kernel is getting interrupts every 5ms and pushing to the iio buffer. The data is timestamped before it's pushed to the buffer.

I have an user space thread that is constantly reading from the iio buffer using iio_buffer_refill(). My buffer is initalized with buffer_samples of 1.

When my user space gets the data, I calculate the latency (TIME_NOW - DATA_TIMESTAMP). I see an latency consistently between 8ms to 12ms. Is this latency expected (It feel it's quite high..)? Where is this latency come from? If there someway to reduce this latency?

rgetz, 2019-07-19:
This is for a local context?

rgetz, 2020-01-13:
Closing, since there was no response. If there still is an issue - please re-open.

closed 2020-01-13 by rgetz