analogdevicesinc/libiio #285: Error while connectiong via network, Unable to create IIO context: Input/output error

zaporozhets, 2019-06-28:
Hi All.

Long story short, I can not connect to iiod daemon via network. 

I see next iiod output on my target machine:

```
# iiod -d -D
Starting IIO Daemon version 0.18

New client connected from 10.42.0.1
Client exited
```
And on my host:
```
#iio_info -u "ip:10.42.0.2"
Library version: 0.15 (git tag: v0.15)
Compiled with backends: local xml ip usb
Unable to create IIO context: Input/output error
```

My target is ZED board with AD-FMCOMMS4 module, kernel from analogdevicesinc repo master branch without modifications. I have tried two versions of libiio(0.16 & 0.18) and got the same error. 

Moreover, everything works fine with prebuilded 2018_R1 image. So might be issue with incompatibility between kernel from master and libiio. What do you think? 

Could you help me with some tips?

Cheers,
Taras.

rgetz, 2019-07-16:
best place for FMCOMMS4 support is ADI's Engineerzone : Just ask the same: https://ez.analog.com/linux-device-drivers/linux-software-drivers

rgetz, 2019-07-20:
closing - if you have any other questions - please ask on Engineerzone.

closed 2019-07-20 by rgetz