analogdevicesinc/libiio #302: Wrong argument help in example.

dimasad, 2019-08-19:
In the examples/dummy-iiostream.c example, the help for the read method is 2 read, 3 read raw, while the order of the enum makes it 2 read raw, 3 read.

rgetz, 2019-08-20:
yeah, it looks like:

https://github.com/analogdevicesinc/libiio/blob/master/examples/dummy-iiostream.c#L159
```
	printf("  -r\tread method (default 0 pointer, 1 callback, 2 read, 3 read raw)\n");
```

and 
https://github.com/analogdevicesinc/libiio/blob/master/examples/dummy-iiostream.c#L94
```
enum {
	BUFFER_POINTER,
	SAMPLE_CALLBACK,
	CHANNEL_READ_RAW,
	CHANNEL_READ,
	MAX_READ_METHOD,
};
```

is in the incorrect order. I will update, and check the others.

Thanks for pointing out.

rgetz, 2020-01-10:
added a fix - should be resolved on the next release.

rgetz, 2020-01-11:
closing this - since the PR/merge is in progress.

Thanks again for pointing this out.

commit bd58f2c6 (2020-01-10): fix #302, wrong argument in usage in dummy-iiostream
commit 2062b1dc (2020-01-11): fix #302, wrong argument in usage in dummy-iiostream
closed 2020-01-11 by rgetz