analogdevicesinc/libiio #303: printf types in dummy-iiostream example not platform independent

dimasad, 2019-08-19:
The examples/dummy-iiostream.c printf types are not platform independent, and do generate compilation error in the armhf architecture, as in the BeagleBone.

rgetz, 2019-08-20:
we build this on Zynq, and cross compile for zynq all time time...

```
root@analog:~# gcc  -v -E -x c++ -dD /dev/null 
Using built-in specs.
COLLECT_GCC=gcc
Target: arm-linux-gnueabihf
```

what errors are you getting?

dimasad, 2019-11-29:
Sorry, I did not get compilation errors, but build-time warnings. When I ran the program, I got weird numbers. Right now I do not have access to the exact messages, but when I get hold of my BBB again I can post it here, if you want.

pcercuei, 2019-12-01:
The ```BUFFER_POINTER``` and ```CHANNEL_READ``` cases uses ```%d``` and ```%ld``` tokens for ```int16_t``` and ```int64_t``` values, they should use ```"%" PRId16``` and ```"%" PRId64``` respectively.

commit f5f99250 (2020-01-13): Merge pull request #304 from dimasad/master

Fixes #303 and #302
closed 2020-01-13 by dNechita