analogdevicesinc/libiio #305: did not performance well on 1000M ethernet

VIRUSOLDLIu, 2019-08-27:
when i use work my devices on 1000M ethernet , the iiod only takes 50% of the cpu and reaches only about 300M ethernet. however when i checked the net using iperf, it works well and reaches about 800M . And iperf takes about 100%of the cpu .  
I also  found that in the iiod it have mutli-stream tool . But I found in my device it only works 1 stream.

rgetz, 2019-08-30:
iio is currently synchronous (buffers are only sent when the host asks). This is a known implementation issue, and something we are working on resolving during libiio next gen discussions.

rgetz, 2020-01-13:
I think this was tested and quantified during a GRCon presentation:
https://www.gnuradio.org/grcon/grcon19/presentations/gr-iio_nuances_hidden_features_and_new_stuff/Travis%20Collins%20-%20gr_iio.pdf

libIIO Performance section (slide 34)

closed 2020-01-13 by rgetz