analogdevicesinc/libiio #307: Discovery of IIO devices with Bonjour on macOS

matejk, 2019-09-02:
As far as I was able to understand, IIOD can broadcast its availability using Avahi.

Client library libiio can discover such devices on Linux using Avahi library.

Since Avahi and Bonjour are compatible, is there a way to use discovery feature in libiio on macOS somehow or shall this part be implemented separately using macOS native APIs?

tfcollins, 2019-09-13:
Auto-discovery works fine out of the box:

```
analogs-mini:~ analog$ uname -a
Darwin analogs-mini.lan 18.7.0 Darwin Kernel Version 18.7.0: Tue Aug 20 16:57:14 PDT 2019; root:xnu-4903.271.2~2/RELEASE_X86_64 x86_64
analogs-mini:~ analog$ iio_attr -u ip:analog -d
IIO context has 6 devices:
	iio:device0: ams, found 1 device attributes
	iio:device1: ad7291, found 0 device attributes
	iio:device2: ad9523-1, found 10 device attributes
	iio:device3: axi-ad9680-hpc, found 0 device attributes
	iio:device4: axi-ad9144-hpc, found 0 device attributes
	iio_sysfs_trigger:, found 2 device attributes
analogs-mini:~ analog$ 
```

matejk, 2019-09-16:
@tfcollins: I see that I wrote misleading text. The devices that I was referring to were IIOD servers on the network, discoverable with Zeroconf.

[image]

libiio has client API to find such servers and I was wondering if it is possible to use that on macOS somehow or shall I use native functions.

tfcollins, 2019-09-16:
This is basically what I showed with my commands. IIOD with autodiscovery is set up on a development board with hostname analog on the network. iio_attr is using the avahi/bonjour to find it.

pcercuei, 2019-09-16:
@tfcollins: you used ```iio_attr -u ip:analog -d```. I believe 'analog' is the local name of the board?

Bonjour discovery works without any name. ```iio_attr -u ip: -d``` should auto-connect to an instance of IIOD running on the local network if Avahi is enabled.

rgetz, 2020-01-03:
```analog``` is the name of the local host of the remote context iiod is running on.

My understanding is: ```iio_attr -u ip:analog -d``` uses the local host DNS to look up the name ```analog.local```. Bonjour/Avhai provides the ```.local``` dns which does a lookup, and provides the IP on the network.

today, I get: 
```
iio_attr -u ip:'' -d
ERROR: Unable to find host: Name or service not known
Unable to create IIO context: No such file or directory
```

but avahi can find things:
```
avahi-browse -d local _ssh._tcp --resolve -t
+ enx00e0229bee20 IPv4 pluto                                         SSH Remote Terminal  local
= enx00e0229bee20 IPv4 pluto                                         SSH Remote Terminal  local
   hostname = [pluto.local]
   address = [192.168.2.1]
   port = [22]
   txt = []
```

pcercuei, 2020-01-10:
I think the solution is just that network_create_context() in network.c should treat an empty "host" parameter as if it was NULL.

rgetz, 2020-01-11:
Yeah, OK.

https://github.com/analogdevicesinc/libiio/blob/master/network.c#L1439

should be something like:
```
     if (!host || host[0] == 0)
```

rgetz, 2020-01-12:
Ok - a little poking - and that is one part, the next is in https://github.com/analogdevicesinc/libiio/blob/master/network.c#L444
```
	browser = avahi_service_browser_new(client,
			AVAHI_IF_UNSPEC, AVAHI_PROTO_UNSPEC,
			"_iio._tcp", NULL, 0, __avahi_browser_cb, &ddata);
```

it is looking for a "service type" of "_iio._tcp", and things like Pluto, don't include that - I looked at /etc/avahi/services and that only includes : 
```
# pwd
/etc/avahi/services
# grep type  *
sftp-ssh.service:    <type>_sftp-ssh._tcp</type>
ssh.service:    <type>_ssh._tcp</type>
```

@pcercuei : does it make sense to include a iio.service somewhere in this repo? (bindings?) or where? (If installing iiod, and avahi is on, then the service file should be installed in /etc/avahi/services?)
@dbogdan and/or @mhennerich ? Thoughts on how this effects the Zynq/Pi SD Cards or Pluto firmware? 

Once I add the file: /etc/avahi/services/iio.service
```
<service-group>

  <name replace-wildcards="yes">%h</name>

  <service>
    <type>_iio._tcp</type>
    <port>30431</port>
  </service>

</service-group>
```
and restart things:
```
# ./S50avahi-daemon reload
```

then (with this change:
```
--- a/network.c
+++ b/network.c
@@ -1436,7 +1436,7 @@ struct iio_context * network_create_context(const char *host)
        hints.ai_socktype = SOCK_STREAM;
 
 #ifdef HAVE_AVAHI
-       if (!host) {
+       if (!host || (host && !host[0])) {
                char addr_str[AVAHI_ADDRESS_STR_MAX];
                char port_str[6];
                AvahiAddress address;
```

things work:

```
rgetz@brain:~/github/libiio/build$ ./tests/iio_attr -u ip: -d
IIO context has 5 devices:
	iio:device0: adm1177, found 0 device attributes
	iio:device1: ad9361-phy, found 18 device attributes
	iio:device2: xadc, found 1 device attributes
	iio:device3: cf-ad9361-dds-core-lpc, found 0 device attributes
	iio:device4: cf-ad9361-lpc, found 0 device attributes
```

I will make a patch, and classify this as a bug.

rgetz, 2020-01-13:
Ok - I'm going to close this, as the fix is in the pull request. @matejk if you want to try things - check that out.

rgetz, 2020-01-21:
I think this fixed things on Linux, but not Mac or Windows :(

[Travis-ci](https://travis-ci.org/analogdevicesinc/libiio/jobs/640107946?utm_medium=notification&utm_source=github_status) and [Appveyor](https://ci.appveyor.com/project/analogdevicesinc/libiio/builds/30266637) says (on both):
```
Building without Avahi (zeroconf) support
```

Which makes sense - Avahi is for Linux only. Linux builds fine : [Travis-ci](https://travis-ci.org/analogdevicesinc/libiio/jobs/640107945?utm_medium=notification&utm_source=github_status) says:
```
 Building with Avahi, a zero-configuration networking (zeroconf) implementation
```

On Windows, and Mac OS, we get Bonjour - Apple's implementation, and it fails our test in `./CMakeLists.txt`
```
        find_library(AVAHI_CLIENT_LIBRARIES avahi-client)
        find_library(AVAHI_COMMON_LIBRARIES avahi-common)
```

I think we need something like [this](https://github.com/deskvox/deskvox/blob/master/cmake/FindBonjour.cmake)

-Robin

Installing on Windows is a pain - (you mostly need to install iTunes, or other apple product). So for the CI parts - I would just look at MAC.

tfcollins, 2020-01-22:
This will require some rewrite of the current zeroconf support since we are using explicit avahi calls which will not work with Bonjour (mDNSResponder) on macOS.

Technically there is a compatibility layer in Avahi to support Bonjour but in their readme they say not to use it: https://www.avahi.org/doxygen/v0.7/html/index.html#choose_api

So implementing the native Bonjour calls is the best way to move forward. Here is the closet thing I could find to an example (which seems to work on Windows too): https://github.com/jevinskie/mDNSResponder/blob/master/Clients/BonjourExample/BonjourExample.cpp

-Travis

rgetz, 2020-01-23:
Ok, just to spec things; we need some development in network.c for MAC and Windows (only).

In the cmake, we need to set a HAVE_BONJOUR (always on for MAC, and one in Windows, if it's installed).

```
#ifdef HAVE_AVAHI
.... <exists, don't touch> ....
#elseif HAVE_BONJOUR
.... <needs to be developed> ....
#endif
```

to implement the same as `discover_host()` ?

As for the example, not to get into a large licensing debate - but we can't really look at that code much - it's under Apache 2.0, which is not compatible with the LGPL2.1+

https://www.apache.org/licenses/GPL-compatibility.html & https://www.gnu.org/licenses/license-list.html#apache2

-Robin

rgetz, 2020-01-23:
I did find this : https://opensource.apple.com/source/mDNSResponder/mDNSResponder-107.5/Clients/BonjourExample/BonjourExample.cpp.auto.html

which is the same file, licensed under the Apple 2.0 license, which is still incompatible with the LGPL2.1+ https://www.gnu.org/licenses/license-list.html#apsl2

So - the answer might be - don't include Bonjour on MAC or Windows...

matejk, 2020-01-24:
It is not necessary to use that source file as it is. New code can be implemented that uses API calls directly and maybe fits better into the existing code of libiio.

On macOS, It is possible to call ObjectiveC API from C code. I might have some time to try that in the next weeks.

rgetz, 2020-01-27:
If you have time to look at things by on a Mac - it would be great. Most of us don’t have access to that hardware to test on...

rgetz, 2020-02-03:
"Discovery of IIO daemons on the network with Bonjour on macOS" is fixed as a bug - with the newly included services file, linux clients will be found.

It's only that libiio isn't calling bonjour to find things itself...

(Which is an enhancement, not a bug).

matejk, 2020-02-22:
I am working on support for DNS-SD (Avahi, Bonjour) for macOS using CFNetServices.

matejk, 2020-02-24:
Implementation of resolving on macOS works. It uses the first IIO service that is provided by the Apple DNS SD API. This behaviour is the same as in the Avahi implementation if I am correct.

Implementing support for multiple IIO services would not be so hard, but I guess this is something for the "next gen IIO"?

I'll run some more tests and prepare PR.

rgetz, 2020-03-31:
Thanks - closing this now, since I think things are all in either master or in PR #412

commit 4a4413bf (2020-01-13): Fix #307 by accepting avahi null names, and install avahi service file

This will make sure things like this are functional:

./tests/iio_attr -u ip: -d
IIO context has 5 devices:
	iio:device0: adm1177, found 0 device attributes
	iio:device1: ad9361-phy, found 18 device attributes
	iio:device2: xadc, found 1 device attributes
	iio:device3: cf-ad9361-dds-core-lpc, found 0 device attributes
	iio:device4: cf-ad9361-lpc, found 0 device attributes
commit 64f9d684 (2020-01-13): Fix #307 by accepting avahi null names, and install avahi service file

This will make sure things like this are functional:

./tests/iio_attr -u ip: -d
IIO context has 5 devices:
	iio:device0: adm1177, found 0 device attributes
	iio:device1: ad9361-phy, found 18 device attributes
	iio:device2: xadc, found 1 device attributes
	iio:device3: cf-ad9361-dds-core-lpc, found 0 device attributes
	iio:device4: cf-ad9361-lpc, found 0 device attributes
commit 673f922b (2020-01-13): Fix #307 by accepting avahi null names, and install avahi service file

This will make sure things like this are functional:

./tests/iio_attr -u ip: -d
IIO context has 5 devices:
	iio:device0: adm1177, found 0 device attributes
	iio:device1: ad9361-phy, found 18 device attributes
	iio:device2: xadc, found 1 device attributes
	iio:device3: cf-ad9361-dds-core-lpc, found 0 device attributes
	iio:device4: cf-ad9361-lpc, found 0 device attributes
commit 0f2445ba (2020-01-14): Merge pull request #338 from analogdevicesinc/rgetz-avahi-install

Fix #307 by accepting avahi null names, and install avahi service file
closed 2020-03-31 by rgetz