analogdevicesinc/libiio #308: Windows (CMake, VS 2017, Python 3.5.1): build of python bindings fails

matejk, 2019-09-05:
When compiling libiio, Python bindings can't be built.

Build fails with an error:

```
CUSTOMBUILD : error : package directory 'C:Users\matejk\Documents\git\libiio\bindings\python' does not exist [C:\Users\matejk\Documents\git\libiio-build-64\bindings\python\libiio-py.vcxproj]
```

The path that can't be found seems to have backslash missing after the C:

libiio.dll and cli tools are built properly.

Environment:

- Windows 10
- Visual Studio 2017
- Build target: Release
- Build platform: x64
- Python version: 3.5.1

Output from CMake

```
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.18362.
-- The C compiler identification is MSVC 19.16.27032.1
-- Check for working C compiler: C:/msvc2017/VC/Tools/MSVC/14.16.27023/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/msvc2017/VC/Tools/MSVC/14.16.27023/bin/Hostx64/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for strdup
-- Looking for strdup - found
-- Looking for strerror_r
-- Looking for strerror_r - not found
-- Looking for newlocale
-- Looking for newlocale - not found
-- Looking for in6addr_any
-- Looking for in6addr_any - not found
CMake Warning at CMakeLists.txt:102 (message):
  IPv6 is not available in your system.

-- Found Git: C:/Users/matejk/AppData/Local/Atlassian/SourceTree/git_local/mingw32/bin/git.exe (found version "2.17.1.windows.2")
-- Looking for libusb-1.0 : Found
-- Looking for libusb_get_version
-- Looking for libusb_get_version - found
-- Found LibXml2: C:/Users/matejk/Documents/git/3rdparty/Windows/x64/lib/libxml2.lib (found version "2.9.9")
-- Found PythonInterp: C:/Python35/python.exe (found version "3.5.1")
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Doxygen not found, API documentation won't be generated
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/matejk/Documents/git/libiio-build-64
```

matejk, 2019-09-17:
Pull request #313 fixes this issue.

rgetz, 2020-01-10:
I think the fixed was merged into master - so closing this issue.

closed 2020-01-10 by rgetz