analogdevicesinc/libiio #309: Windows (CMake, VS 2017): install of c-sharp bindings fails

matejk, 2019-09-05:
When installing libiio, c-sharp bindings can't be built.

Build fails with an error:

```
  -- Installing: C:/Users/matejk/Documents/git/Windows/x64/lib/cli/libiio-sharp-0.18/libiio-sharp.dll
  CMake Error at bindings/csharp/cmake_install.cmake:68 (file):
    file INSTALL cannot find
    "C:/Users/matejk/Documents/git/libiio-build-64/bindings/csharp/libiio-sharp.dll.mdb".
  Call Stack (most recent call first):
    bindings/cmake_install.cmake:37 (include)
    cmake_install.cmake:69 (include)
```

libiio.dll and cli tools are built properly.

Environment:

- Windows 10
- Visual Studio 2017
- Build target: Release
- Build platform: x64

Complete relevant part of output:

```
  Generating libiio-sharp.dll
  Microsoft (R) Visual C# Compiler version 2.10.0.0 (b9fb1610)
  Copyright (C) Microsoft Corporation. All rights reserved.

CUSTOMBUILD : warning CS1668: Invalid search path 'lib\um\x64' specified in 'LIB environment variable' -- 'directory does not exist' [C:\Users\matejk\Documents\git\libiio-build-64\bindings\csharp\libiio-sharp.vcxproj]
CUSTOMBUILD : warning CS1668: Invalid search path 'Lib\um\x64' specified in 'LIB environment variable' -- 'directory does not exist' [C:\Users\matejk\Documents\git\libiio-build-64\bindings\csharp\libiio-sharp.vcxproj]
  Building Custom Rule C:/Users/matejk/Documents/git/libiio/bindings/csharp/CMakeLists.txt
  Building Custom Rule C:/Users/matejk/Documents/git/libiio/CMakeLists.txt
  -- Install configuration: "Release"
  -- Installing: C:/Users/matejk/Documents/git/Windows/x64/lib/pkgconfig/libiio.pc
  -- Installing: C:/Users/matejk/Documents/git/Windows/x64/lib/libiio.lib
  -- Installing: C:/Users/matejk/Documents/git/Windows/x64/bin/libiio.dll
  -- Up-to-date: C:/Users/matejk/Documents/git/Windows/x64/include/iio.h
  -- Installing: C:/Users/matejk/Documents/git/Windows/x64/lib/pkgconfig/libiio-sharp-0.18.pc
  -- Installing: C:/Users/matejk/Documents/git/Windows/x64/lib/cli/libiio-sharp-0.18/libiio-sharp.dll.config
  -- Installing: C:/Users/matejk/Documents/git/Windows/x64/lib/cli/libiio-sharp-0.18/libiio-sharp.dll
  CMake Error at bindings/csharp/cmake_install.cmake:68 (file):
    file INSTALL cannot find
    "C:/Users/matejk/Documents/git/libiio-build-64/bindings/csharp/libiio-sharp.dll.mdb".
  Call Stack (most recent call first):
    bindings/cmake_install.cmake:37 (include)
    cmake_install.cmake:69 (include)

C:\msvc2017\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: The command "setlocal [C:\Users\matejk\Documents\git\libiio-build-64\install.vcxproj]
C:\msvc2017\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: "C:\Program Files\CMake\bin\cmake.exe" -DBUILD_TYPE=Release -P cmake_install.cmake [C:\Users\matejk\Documents\git\libiio-build-64\install.vcxproj]
C:\msvc2017\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: if %errorlevel% neq 0 goto :cmEnd [C:\Users\matejk\Documents\git\libiio-build-64\install.vcxproj]
C:\msvc2017\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: :cmEnd [C:\Users\matejk\Documents\git\libiio-build-64\install.vcxproj]
C:\msvc2017\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: endlocal & call :cmErrorLevel %errorlevel% & goto :cmDone [C:\Users\matejk\Documents\git\libiio-build-64\install.vcxproj]
C:\msvc2017\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: :cmErrorLevel [C:\Users\matejk\Documents\git\libiio-build-64\install.vcxproj]
C:\msvc2017\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: exit /b %1 [C:\Users\matejk\Documents\git\libiio-build-64\install.vcxproj]
C:\msvc2017\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: :cmDone [C:\Users\matejk\Documents\git\libiio-build-64\install.vcxproj]
C:\msvc2017\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: if %errorlevel% neq 0 goto :VCEnd [C:\Users\matejk\Documents\git\libiio-build-64\install.vcxproj]
C:\msvc2017\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: :VCEnd" exited with code 1. [C:\Users\matejk\Documents\git\libiio-build-64\install.vcxproj]
```

rgetz, 2020-01-10:
@AlexandraTrifan : can you confirm this is fixed now?

AlexandraTrifan, 2020-01-13:
Hi,

I just tested the build using CMake-GUI to configure and generate the CMake project. Then opened the project in Visual Studio and it finished the build successfully.
@matejk Does this still happen to you? 
Can you tell us the C# compiler version that is being used? This can be checked in the CMake configuration under the variable MCS_EXECUTABLE. (For me it was: C:/Windows/Microsoft.NET/Framework/v4.0.30319/csc.exe ). 

Thank you,
-Alexandra

matejk, 2020-01-13:
Hi @AlexandraTrifan. The version of C# is 
```
C:/Windows/Microsoft.NET/Framework/v4.0.30319/csc.exe
```
Note that the binding compiles fine. The install step fails because it expects that the following file exists:

```
bindings/csharp/libiio-sharp.dll.mdb
```

AlexandraTrifan, 2020-01-13:
Hi, 

Sorry for the inconvenience.
I found the issue and added a fix for it which can be tracked in PR #337 . 
Will be available on master after review.

Thank you, 
-Alexandra

rgetz, 2020-01-13:
@AlexandraTrifan : Thanks

rgetz, 2020-01-13:
@matejk : you should be able to try out the pull request artifacts at:
https://ci.appveyor.com/project/analogdevicesinc/libiio/builds/30077854/artifacts

to check things out.

matejk, 2020-01-14:
I confirm that IIO C# bindings from branch `fix_install_csharp` build and install properly with VS 2017.

rgetz, 2020-01-14:
Thanks for the confirmation (and report) - I will close this and merge the pull request.

closed 2020-01-14 by rgetz