analogdevicesinc/libiio #310: Build for libiio (> 0.15) fails on PetaLinux 2018.3: contains probably-redundant RPATH /usr/lib [useless-rpaths]

matejk, 2019-09-05:
Latest versions of libiio can't be built on PetaLinux 2018.3 (for target ultra96-zynqmp).

Version 0.15 is the last that builds without the error.

Build fails with error "contains probably-redundant RPATH /usr/lib [useless-rpaths]":

```
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: libiio-0.18-gitAUTOINC+4e22517c60-r0 do_package_qa: QA Issue: libiio: /work/aarch64-xilinx-linux/libiio/0.18-gitAUTOINC+4e22517c60-r0/packages-split/libiio/usr/lib/libiio.so.0.18 contains probably-redundant RPATH /usr/lib [useless-rpaths]
ERROR: libiio-0.18-gitAUTOINC+4e22517c60-r0 do_package_qa: QA Issue: libiio-iiod: /work/aarch64-xilinx-linux/libiio/0.18-gitAUTOINC+4e22517c60-r0/packages-split/libiio-iiod/usr/sbin/iiod contains probably-redundant RPATH /usr/lib [useless-rpaths]
ERROR: libiio-0.18-gitAUTOINC+4e22517c60-r0 do_package_qa: QA Issue: libiio-tests: /work/aarch64-xilinx-linux/libiio/0.18-gitAUTOINC+4e22517c60-r0/packages-split/libiio-tests/usr/bin/iio_reg contains probably-redundant RPATH /usr/lib
libiio-tests: /work/aarch64-xilinx-linux/libiio/0.18-gitAUTOINC+4e22517c60-r0/packages-split/libiio-tests/usr/bin/iio_info contains probably-redundant RPATH /usr/lib
libiio-tests: /work/aarch64-xilinx-linux/libiio/0.18-gitAUTOINC+4e22517c60-r0/packages-split/libiio-tests/usr/bin/iio_adi_xflow_check contains probably-redundant RPATH /usr/lib
libiio-tests: /work/aarch64-xilinx-linux/libiio/0.18-gitAUTOINC+4e22517c60-r0/packages-split/libiio-tests/usr/bin/iio_readdev contains probably-redundant RPATH /usr/lib
libiio-tests: /work/aarch64-xilinx-linux/libiio/0.18-gitAUTOINC+4e22517c60-r0/packages-split/libiio-tests/usr/bin/iio_genxml contains probably-redundant RPATH /usr/lib
libiio-tests: /work/aarch64-xilinx-linux/libiio/0.18-gitAUTOINC+4e22517c60-r0/packages-split/libiio-tests/usr/bin/iio_writedev contains probably-redundant RPATH /usr/lib
libiio-tests: /work/aarch64-xilinx-linux/libiio/0.18-gitAUTOINC+4e22517c60-r0/packages-split/libiio-tests/usr/bin/iio_attr contains probably-redundant RPATH /usr/lib [useless-rpaths]
ERROR: libiio-0.18-gitAUTOINC+4e22517c60-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: libiio-0.18-gitAUTOINC+4e22517c60-r0 do_package_qa: Function failed: do_package_qa
ERROR: Logfile of failure stored in: /home/matejk/us-software/petalinux-2018.3-ultra96/build/tmp/work/aarch64-xilinx-linux/libiio/0.18-gitAUTOINC+4e22517c60-r0/temp/log.do_package_qa.6897
ERROR: Task (/home/matejk/us-software/petalinux-2018.3-ultra96/project-spec/meta-user/recipes-apps/libiio/libiio.bb:do_package_qa) failed with exit code '1'
```

I use [bitbake from openembeded](http://cgit.openembedded.org/meta-openembedded/tree/meta-oe/recipes-support/libiio/libiio_git.bb?h=master) with slight modification to install Sys V init service.

```
#
# Modelled from: https://github.com/openembedded/meta-openembedded/blob/master/meta-oe/recipes-support/libiio/libiio_git.bb
#

SUMMARY = "Analog Devices Industrial IO library"
HOMEPAGE = "https://wiki.analog.com/resources/tools-software/linux-software/libiio"
SECTION = "PETALINUX/apps"
LICENSE = "LGPLv2.1+"
LIC_FILES_CHKSUM = "file://COPYING.txt;md5=7c13b3376cea0ce68d2d2da0a1b3a72c"

# 0.18
SRCREV = "4e22517c60f3c5e691320871956edede15459ae3"
IIOVER="0.18"

# 0.15
#SRCREV = "6ecff5d46e1b12c2859f0b63a73282940e3402bb"
#IIOVER="0.15"

# 0.10
#SRCREV = "11b871b4f5fb1b9e9029fae1a7fd3734efb441b0"
#IIOVER="0.10"

SRC_URI = "git://github.com/analogdevicesinc/libiio.git \
           file://iiod.sysv"

S = "${WORKDIR}/git"

#inherit cmake gitpkgv systemd update-rc.d
inherit cmake gitpkgv update-rc.d

PV = "${IIOVER}-git${SRCPV}"
PKGV = "${IIOVER}-git${GITPKGV}"

DEPENDS = " \
    flex-native bison-native libaio \
    ${@bb.utils.contains('DISTRO_FEATURES', 'zeroconf', 'avahi', '', d)} \
"

EXTRA_OECMAKE = " \
    -DCMAKE_BUILD_TYPE=RelWithDebInfo \
    -DUDEV_RULES_INSTALL_DIR=${nonarch_base_libdir}/udev/rules.d \
    -DWITH_IIOD=ON -DWITH_IIOD_CLIENT=ON \
    ${@bb.utils.contains('DISTRO_FEATURES', 'systemd', '-DWITH_SYSTEMD=ON', '', d)} \
    ${@bb.utils.contains('DISTRO_FEATURES', 'sysvinit', '-DWITH_SYSVINIT=ON', '', d)} \
"

PACKAGECONFIG ??= "USB_BACKEND NETWORK_BACKEND PYTHON_BINDINGS"

PACKAGECONFIG[USB_BACKEND] = "-DWITH_USB_BACKEND=ON,-DWITH_USB_BACKEND=OFF,libusb1,libxml2"
PACKAGECONFIG[NETWORK_BACKEND] = "-DWITH_NETWORK_BACKEND=ON,-DWITH_NETWORK_BACKEND=OFF,libxml2"
PACKAGECONFIG[PYTHON_BINDINGS] = ",,python"

PACKAGES =+ "${PN}-iiod ${PN}-tests ${PN}-python"

# Corrected IIOD sysvinit script
do_install_append () {
        rm ${D}/${sysconfdir}/init.d/iiod
        install -m 0755 ${WORKDIR}/iiod.sysv ${D}/${sysconfdir}/init.d/iiod
}

FILES_${PN}-iiod = " \
    ${sbindir}/iiod \
    ${systemd_system_unitdir}/iiod.service \
    ${sysconfdir}/init.d/iiod \
"
FILES_${PN}-tests = "${bindir}"
FILES_${PN}-python = "${PYTHON_SITEPACKAGES_DIR}"

#SYSTEMD_PACKAGES = "${PN}-iiod"
#SYSTEMD_SERVICE_${PN}-iiod = "iiod.service"

INITSCRIPT_PACKAGES = "${PN}-iiod"
INITSCRIPT_NAME_${PN}-iiod = "iiod"
INITSCRIPT_PARAMS_${PN}-iiod = "defaults"

IMAGE_INSTALL_append = " libiio libiio-iiod"
```

mnsgs, 2019-09-05:
This is resolved through https://github.com/analogdevicesinc/libiio/commit/5090603d01779bb1717fb0c50953330e8770550f

matejk, 2019-09-06:
I can confirm that merge request #283 solves this problem.

closed 2019-09-06 by matejk