analogdevicesinc/libiio #311: Use space chars in iio.py

eblot, 2019-09-16:
It would be nice to follow PEP8 and replace tabs with space chars in `iio.py`

https://www.python.org/dev/peps/pep-0008/#tabs-or-spaces

pcercuei, 2019-09-16:
We can't really change that now, because it would trash the whole git history of the file... (you wouldn't be able to git-blame afterwards)

julienmalik, 2019-09-17:
Reading https://eev.ee/blog/2016/06/04/converting-a-git-repo-from-tabs-to-spaces/ :

`git blame -w` ignores whitespace only changes

my 2 cents...

rgetz, 2020-01-13:
Yeah, command line git blame, most git UI support it...
https://github.com/gitextensions/gitextensions/issues/4915

the html interface (github) does not (that I can tell).

pcercuei, 2020-01-15:
Well PEB8 only says that spaces are the *preferred* indentation method, and that we shouldn't mix spaces with tabs, which we don't.

eblot, 2020-01-15:
It is not mandatory you are right. However tab usage is quite sparse with Python community (ex: https://ukupat.github.io/tabs-or-spaces/), so it is definitely not easy to work with Python projects where pretty much all Python modules follow the same (space) rule except `libiio`. 
It is not a big deal, but it definitely does not help to write patches for it, IMHO.

rgetz, 2020-01-29:
yeah, I ran into a problem with vim recently - any file with the suffix of ".py" it assumes it wants to be indented with spaces. I needed to rename the file in order it would keep the tabs....

Found this [here](https://www.i-programmer.info/news/99-professional/10049-tabs-or-spaces-one-billion-files-later-an-answer.html) Spaces "win" for python.
[image]

pcercuei, 2020-01-30:
Keep in mind you need 4 or 8 spaces for one tab ;)

rgetz, 2020-02-02:
@pcercuei : it's 184 lines, so, the file size would increase by 552 bytes... (tabs would be removed, and replaced by 4 spaces). I think that is something we can do (if we want) :)

rgetz, 2020-03-31:
I am in process of adding black/flake-8 processing as part of PR #413 - so we will track things there...

If you have any comments - please comment there.

closed 2020-03-31 by rgetz