analogdevicesinc/libiio #315: Units of channel values acquired using libiio

matejk, 2019-09-19:
What are the units used in the IIO infrastructure after converting channel raw values (applying offset, scale, ..)?

Does IIO prescribe the units?

Are milli units (mV, mA, mC) a convention?

Does a list of units for each channel types (`enum iio_chan_type`) exist already?

The only information that I was able to find is on the page of [ADIS16400 IIO Inertial Measurement Unit Linux Driver](https://wiki.analog.com/resources/tools-software/linux-drivers/iio-inertial-measurement-units/adis16400#show_channel_value):

> The units by the IIO framework are:
> 
> - Acceleration: Meter per Second squared
> - Angular velocity: Rad per second
> - Pressure: kilo Pascal
> - Voltage: milli Volts
> - Temperature: milli-degree Celsius

rgetz, 2020-01-03:
It depends on what things are.

Nominally everything is in milli SI units (since no floating point in the kernel)
- voltage : millivolts
- current : milliamps
- temperature : milli degrees Celsius

Note that this will not include voltage dividers or amplifiers you may have on the inputs of ADCs.

rgetz, 2020-01-10:
Close since I think this is answered. If you still have questions, please re-open

closed 2020-01-10 by rgetz