analogdevicesinc/libiio #317: Can't find libxml in Linux

db4rne, 2019-10-11:
while building fro source, libiio seems to be unable to find libxml2, at least unter arch linux.

this is the output if `make all` after cloning the repo and `cmake ./`:

```
$ make all 
Scanning dependencies of target iio
[  2%] Building C object CMakeFiles/iio.dir/backend.c.o
[  5%] Building C object CMakeFiles/iio.dir/channel.c.o
[  7%] Building C object CMakeFiles/iio.dir/device.c.o
[ 10%] Building C object CMakeFiles/iio.dir/context.c.o
[ 12%] Building C object CMakeFiles/iio.dir/buffer.c.o
[ 15%] Building C object CMakeFiles/iio.dir/utilities.c.o
[ 17%] Building C object CMakeFiles/iio.dir/scan.c.o
[ 20%] Building C object CMakeFiles/iio.dir/sort.c.o
[ 22%] Building C object CMakeFiles/iio.dir/local.c.o
[ 25%] Building C object CMakeFiles/iio.dir/usb.c.o
[ 27%] Building C object CMakeFiles/iio.dir/serial.c.o
[ 30%] Building C object CMakeFiles/iio.dir/network.c.o
[ 32%] Building C object CMakeFiles/iio.dir/xml.c.o
/home/arne/tmp/libiio/xml.c:23:10: fatal error: libxml/tree.h: No such file or directory
   23 | #include <libxml/tree.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/iio.dir/build.make:219: CMakeFiles/iio.dir/xml.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:79: CMakeFiles/iio.dir/all] Error 2
make: *** [Makefile:130: all] Error 
```

I would expect it to build here.

pcercuei, 2019-10-12:
Well, did you install the development files of libxml2?

db4rne, 2019-10-12:
Yes, the *-dev packages don't exist in Arch Linux, so I can't forget it. Also: shouldn't not finding libxml already throw an error in cmake?

```
$ pacman -Ql libxml2
[...]
libxml2 /usr/include/libxml2/libxml/tree.h
[...]
```

tfcollins, 2019-10-13:
Worked fine on my manjaro machine with just the base-devel group installed.

pcercuei, 2019-10-14:
So what include directory does CMake uses for libxml2? It should be ```/usr/include/libxml2```

tfcollins, 2019-10-14:
From cmake:
```
LIBXML2_INCLUDE_DIRS: /usr/include/usr/include/libxml2/usr/include/usr/include
LIBXML2_LIBRARIES: -L/usr/lib-lxml2-lpthread/usr/lib/liblzma.so/usr/lib/libz.so-licui18n-licuuc-licudata-lm
```

tfcollins, 2019-10-17:
@pcercuei correct me if I'm wrong here, but we don't really do anything special with libxml2 in cmake but rely on find_package.

My cmake generated path in `iiod/CMakeFiles/iiod.dir/DependInfo.cmake` includes `/usr/inlucde/libxml2`

I have libxml2 version 2.9.9-2 installed with cmake 3.15.4 on my Manjaro system.

Even the AUR version doesn't do anything weird in the PKGBUILD

db4rne, 2019-10-31:
I really don't get why it's not building on my systems, my workaround now is to set it to REQUIRED in the CMakeList.txt file, but I honestly have no idea what exactly that changes, it just works now. I got the idea from the LibXML2 documentation.

Here is the diff in case someone else has the same problem

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 634e553..53c623a 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -255,7 +255,7 @@ endif()
 # So, try first to find the CMake module provided by libxml2 package, then fallback
 # on the CMake's FindLibXml2.cmake module (which can lack some definition, especially
 # in static build case).
-find_package(LibXml2 QUIET NO_MODULE)
+find_package(LibXml2 REQUIRED)
 if(DEFINED LIBXML2_VERSION_STRING)
        set(LIBXML2_FOUND ON)
        set(LIBXML2_INCLUDE_DIR ${LIBXML2_INCLUDE_DIRS})
```

what didn't work, but I also tried: building and installing AUR version libxml2-git as well as installing libxml2-2.9.9-2

paholden, 2020-01-02:
Was there any resolution to this problem?  I am also running into the failure when trying to cross-compile libraries.

db4rne, 2020-01-02:
Hi,

sorry for going dark on this one, I realized it was probably a problem with cmake or some other component.

long story short: if "/bin" is in your PATH, cmake doesn't work correctly. I don't (yet) know why, but I found out what is putting that in my PATH in the first place: xdm.

Working PATH (for me): `/usr/local/sbin:/usr/local/bin:/usr/bin:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl`

paholden, 2020-01-02:
Thanks, I'll play around with my PATH and see if I can get it going.

paholden, 2020-01-02:
I've tried to run with & without "/bin" in the PATH and get the same results.  xml.c always uses /usr/include even though I set my own sysroot path.

Command line sequence for building libiio:
mkdir build
cd build
cmake --sysroot=${PWD}/../../ubuntu-18.04.2-minimal-armhf-2019-02-16 ../
make all

[  3%] Building C object CMakeFiles/iio.dir/backend.c.o
[  7%] Building C object CMakeFiles/iio.dir/channel.c.o
[ 11%] Building C object CMakeFiles/iio.dir/device.c.o
[ 15%] Building C object CMakeFiles/iio.dir/context.c.o
[ 19%] Building C object CMakeFiles/iio.dir/buffer.c.o
[ 23%] Building C object CMakeFiles/iio.dir/utilities.c.o
[ 26%] Building C object CMakeFiles/iio.dir/scan.c.o
[ 30%] Building C object CMakeFiles/iio.dir/sort.c.o
[ 34%] Building C object CMakeFiles/iio.dir/local.c.o
[ 38%] Building C object CMakeFiles/iio.dir/network.c.o
[ 42%] Building C object CMakeFiles/iio.dir/xml.c.o
In file included from /usr/include/libxml2/libxml/parser.h:810:0,
                 from /usr/include/libxml2/libxml/globals.h:18,
                 from /usr/include/libxml2/libxml/threads.h:35,
                 from /usr/include/libxml2/libxml/xmlmemory.h:218,
                 from /usr/include/libxml2/libxml/tree.h:1307,
                 from /home/patricaholden/williamson/testbuild/Pro3-Linux/packages/libiio/xml.c:23:
/usr/include/libxml2/libxml/encoding.h:31:26: fatal error: unicode/ucnv.h: No such file or directory
 #include <unicode/ucnv.h>
                          ^
compilation terminated.
CMakeFiles/iio.dir/build.make:302: recipe for target 'CMakeFiles/iio.dir/xml.c.o' failed
make[2]: *** [CMakeFiles/iio.dir/xml.c.o] Error 1

paholden, 2020-01-03:
More i
nfo:

I started with a fresh copy of the LibIIO repo and re-ran cmake.  Here are my steps and the results.  As you can see, libxml2 is being found in /usr/lib and not in the sysroot defined.  I suspect that libusb is also being found in /usr/lib.  So I'm not sure if this correct behaviour, since the libxml & libusb are both in the sysroot specified in the cmake line.  I can build libiio using "make all", but then "make install" fails trying to write to /usr/lib even though I've specified my own sysroot in the cmake line.  I have checked the pathname for sysroot and it is most certainly correct.

git clone  https://github.com/analogdevicesinc/libiio.git
cd libiio
mkdir build
cd build
cmake --sysroot=${PWD}/../../../ubuntu-18.04.2-minimal-armhf-2019-02-16 ../
-- The C compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test HAS_WPEDANTIC
-- Performing Test HAS_WPEDANTIC - Success
-- Looking for strdup
-- Looking for strdup - found
-- Looking for strerror_r
-- Looking for strerror_r - found
-- Looking for newlocale
-- Looking for newlocale - found
-- Looking for in6addr_any
-- Looking for in6addr_any - found
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- Looking for libusb-1.0 : Found
-- Looking for libusb_get_version
-- Looking for libusb_get_version - found
-- Performing Test HAS_O_TMPFILE
-- Performing Test HAS_O_TMPFILE - Success
-- Performing Test WITH_NETWORK_EVENTFD
-- Performing Test WITH_NETWORK_EVENTFD - Success
-- Performing Test HAS_PIPE2
-- Performing Test HAS_PIPE2 - Success
-- Found LibXml2: /usr/lib/x86_64-linux-gnu/libxml2.so (found version "2.9.4") 
CMake Warning at bindings/python/CMakeLists.txt:4 (find_package):
  By not providing "FindPython.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Python", but
  CMake did not find one.

  Could not find a package configuration file provided by "Python" with any
  of the following names:

    PythonConfig.cmake
    python-config.cmake

  Add the installation prefix of "Python" to CMAKE_PREFIX_PATH or set
  "Python_DIR" to a directory containing one of the above files.  If "Python"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Doxygen not found, API documentation won't be generated
-- Found BISON: /usr/bin/bison (found version "3.0.4") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of struct usb_functionfs_descs_head_v2
-- Check size of struct usb_functionfs_descs_head_v2 - done
-- Looking for pthread_setname_np
-- Looking for pthread_setname_np - found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/patricaholden/williamson/testbuild/Pro3-Linux/packages/libiio/build

rgetz, 2020-01-03:
Just to point out - we already provide binaries for armhf (Raspbian Stretch & Raspbian Jessie), they are linked from the readme at:
https://github.com/analogdevicesinc/libiio/blob/master/README.md

paholden, 2020-01-03:
Thanks for pointing this out as I've kind of been stuck on this for a bit trying to do it from CMake within my project.  Will check out your README and see if this fits in my project scope, which I think it should.

rgetz, 2020-01-10:
I'm going to close this - as I think the cross problem is in a new thread.

closed 2020-01-10 by rgetz