analogdevicesinc/libiio #318: OSX High Sierra, Unable to create IIO context: Function not implemented

cpbridges, 2019-10-12:
Hi Everyone,

Apologies if this is basic. I've tried compiling 1/ from src, 2/ via macports, and 3/ via the prebuilt installer. I keep coming up against this error:

```
Chris-MacBookAir:build chrisbridges$ iio.framework/Tools/iio_info 
Library version: 0.18 (git tag: c0012d0)
Compiled with backends: xml ip usb
Unable to create IIO context: Function not implemented
```

Any thoughts? 73s Chris

tfcollins, 2019-10-12:
You need to provide arguments to iio_info unless you have local IIO devices. Try: `iio_info -s`

cpbridges, 2019-10-12:
Ugh. What a noob - thanks Travis.

```
(py37) Chris-MacBookAir:Work chrisbridges$ iio_info -s
Library version: 0.18 (git tag: v0.18)
Compiled with backends: xml ip usb
Available contexts:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=104400b8399100180d0004001f8a65f6d6 [usb:250.5.5]
```

closed 2019-10-12 by cpbridges