analogdevicesinc/libiio #324: Why do you align the buffer pointer to a multiple of the size of the current scan element in iio_buffer_foreach_sample()?

lmoellendorf, 2019-11-25:
# Description

I have written a custom kernel module implementing the IIO device API backed by an IIO triggered buffer.

My IIO device provides 3 channels + timestamp. The sizes of the channels are

index | iio_chan_spec.scan_type.storagebits
-------|------------------------------------------------
   0   |  32
   1   |  16
   2   |  16

## Kernelspace

From the [description of iio_push_to_buffers_with_timestamp()]:

> The sample data buffer needs to be large enough to hold the additional timestamp (usually the buffer should be indio->scan_bytes bytes large).

In my kernel module I implement [iio_info.update_scan_mode](https://elixir.bootlin.com/linux/v4.9.87/source/include/linux/iio/iio.h#L435) to  create a temporary sample data buffer for my scan elements of size [iio_dev.scan_bytes](https://elixir.bootlin.com/linux/v4.9.87/source/include/linux/iio/iio.h#L518).

In my pollfunc bottom half I store my samples into this temporary buffer and pass a pointer to it to [iio_push_to_buffers_with_timestamp()](https://elixir.bootlin.com/linux/v4.9.87/source/include/linux/iio/buffer.h#L160).

## Userspace

In userspace I am using [iio_buffer_foreach_sample](https://analogdevicesinc.github.io/libiio/group__Buffer.html#ga8be1b9fa5383746ff8d76cedf1ff0156) to read my data scans.

If I enable all 3 channels everything works as expected. But if I only select channel 0 (32bit) and one of 1 or 2 (each 16bit), I read garbage for channel 0 at the second iteration of `iio_buffer_foreach_sample()`. The reason is [line 229f of iio_buffer_foreach_sample()](https://github.com/analogdevicesinc/libiio/blob/master/buffer.c#L229):

```c
			if ((ptr - start) % length)
				ptr += length - ((ptr - start) % length);

```

On first iteration:
-  ptr points to channel 0 and after processing...
- ptr is moved by 4 bytes to channel 1 (or channel 2 depending on which channel is selected) and after processing...
- ptr is moved by 2 bytes (the length of channel 1 or 2)

On second iteration:

-  ptr is moved by another 2 bytes to align its address to a multiple of the size of channel 0.

But this is absolutely wrong in my case. 

If I comment out line 229 and 230 everything works as expected.

# Questions

- Why are you aligning the pointer at all?
- If you have a good reason for aligning the pointer, what am I doing wrong in my driver?

pcercuei, 2019-11-25:
Hi Lars,

If you sample one 32-bit channel and one 16-bit channel, you must add 16 bits of padding at the end of the sample so that the next 32-bit value lies on a 32-bit boundary. This is required since a lot of architectures cannot do unaligned memory access.

lmoellendorf, 2019-11-26:
> If you sample one 32-bit channel and one 16-bit channel, you must add 16 bits of padding at the end of the sample so that the next 32-bit value lies on a 32-bit boundary.

If I select channel 0 (32bit) one of channel 1 or 2 (16bit) [indio_dev->scan_bytes](https://elixir.bootlin.com/linux/v4.9.87/source/include/linux/iio/iio.h#L518) which is used in [iio_push_to_buffers_with_timestamp](https://elixir.bootlin.com/linux/v4.9.87/source/include/linux/iio/buffer.h#L164) to calculate the offset of the timestamp and [buffer.bytes_per_datum](https://elixir.bootlin.com/linux/v4.9.87/source/include/linux/iio/buffer.h#L100) which determines the size of an single kfifo element have a value of 6Byte which does not account for any alignment. Do  have to overwrite them manually? Where?

On the other hand I see a call to [iio_demux](https://elixir.bootlin.com/linux/v4.9.87/source/drivers/iio/industrialio-buffer.c#L1231) in [iio_push_to_buffer](https://elixir.bootlin.com/linux/v4.9.87/source/drivers/iio/industrialio-buffer.c#L1247) which looks like it would do what I need. However, in my driver `buffer->demux_list` stays empty. What do I have to do to get this working? Call  `iio_update_buffers()` which in turn calls `iio_update_demux()` -> [iio_buffer_update_demux](https://elixir.bootlin.com/linux/v4.9.87/source/drivers/iio/industrialio-buffer.c#L1308)  [here](https://elixir.bootlin.com/linux/v4.9.87/source/drivers/iio/industrialio-buffer.c#L765)? Or call `iio_update_demux()` directly? Where?

And if I sample one 32-bit channel and one 64-bit channel (the timestamp), do I have to pad the 32-bit sample as well to align it with 64-bit memory access?

pcercuei, 2019-11-26:
The scan_bytes is updated when the buffer is enabled. See: https://elixir.bootlin.com/linux/latest/source/drivers/iio/industrialio-buffer.c#L565 and its callers.

If you sample one 32-bit channel and one 64-bit channel, then you must pad the 32-bit sample to respect the alignment, that's correct.

lmoellendorf, 2019-11-26:
After having a closer look at  [iio_compute_scan_bytes](https://elixir.bootlin.com/linux/v4.9.87/source/drivers/iio/industrialio-buffer.c#L545) which is responsible for calculating both, `indio_dev.scan_bytes` and `iio_buffer.bytes_per_datum` found out that the order of channels matter: 

```c
	/* How much space will the demuxed element take? */
	for_each_set_bit(i, mask,
			 indio_dev->masklength) {
		length = iio_storage_bytes_for_si(indio_dev, i);
		bytes = ALIGN(bytes, length);
		bytes += length;
	}
```
In case the length of each scan element is smaller then the length of the successive scan elements this algorithm works.
In my case the length of channel 0's scan elements  is greater then the size of the samples of the consecutive channels so no alignment is taken into account.
To make the algorithm work for any case the greatest length of all enabled scan elements has to be used for alignment, e.g.:

```diff
diff --git a/drivers/iio/industrialio-buffer.c b/drivers/iio/industrialio-buffer.c
index 5d05c38c4ba9..3d2c4e26d818 100644
--- a/drivers/iio/industrialio-buffer.c
+++ b/drivers/iio/industrialio-buffer.c
@@ -546,16 +546,18 @@ static int iio_compute_scan_bytes(struct iio_dev *indio_dev,
 				const unsigned long *mask, bool timestamp)
 {
 	unsigned bytes = 0;
-	int length, i;
+	int length, i, align = 0, count = 0;
 
 	/* How much space will the demuxed element take? */
 	for_each_set_bit(i, mask,
 			 indio_dev->masklength) {
 		length = iio_storage_bytes_for_si(indio_dev, i);
-		bytes = ALIGN(bytes, length);
-		bytes += length;
+		align = max(align, length);
+		count++;
 	}
 
+	bytes = count * align;
+
 	if (timestamp) {
 		length = iio_storage_bytes_for_timestamp(indio_dev);
 		bytes = ALIGN(bytes, length);

```

And if the storage bytes for timestamp have to be taken into account for the alignment of the scan elements (what is not the case as it is implemented currently), then:

```diff
diff --git a/drivers/iio/industrialio-buffer.c b/drivers/iio/industrialio-buffer.c
index 5d05c38c4ba9..59aee2ea4e19 100644
--- a/drivers/iio/industrialio-buffer.c
+++ b/drivers/iio/industrialio-buffer.c
@@ -546,21 +546,23 @@ static int iio_compute_scan_bytes(struct iio_dev *indio_dev,
 				const unsigned long *mask, bool timestamp)
 {
 	unsigned bytes = 0;
-	int length, i;
+	int length, i, align = 0, count = 0;
+
+	if (timestamp) {
+		align = iio_storage_bytes_for_timestamp(indio_dev);
+		count++;
+	}
 
 	/* How much space will the demuxed element take? */
 	for_each_set_bit(i, mask,
 			 indio_dev->masklength) {
 		length = iio_storage_bytes_for_si(indio_dev, i);
-		bytes = ALIGN(bytes, length);
-		bytes += length;
+		align = max(align, length);
+		count++;
 	}
 
-	if (timestamp) {
-		length = iio_storage_bytes_for_timestamp(indio_dev);
-		bytes = ALIGN(bytes, length);
-		bytes += length;
-	}
+	bytes = count * align;
+
 	return bytes;
 }
```
 
But in my case the latter would bloat the buffer from 16 Byte to 4*16 = 64 Byte per sample if all channels are selected and timestamp is active.

For now, I will work around this by using 32 storagebits for all my channels. This gives my 4 Bytes of overhead per sample if all elements are selected and additional 2 Byte if timestamp is active.

pcercuei, 2019-11-26:
I think the current algorithm is correct, but the value returned by ```iio_compute_scan_bytes``` should account for the padding. So it should return 8 bytes, not 6 bytes in your case.

I suggest that you bring this issue on the Linux kernel's IIO mailing list.

rgetz, 2020-01-13:
I'm going to close this, assuming it's moved to the iio mailing list at:
https://marc.info/?l=linux-iio&m=157503820832052&w=2

as this does seem to be a kernel issue

closed 2020-01-13 by rgetz