analogdevicesinc/libiio #329: [Announcement] Rust language wrapper for libiio

fpagliughi, 2019-12-29:
Hey All. Apologies if this isn't the best place to post this, but I wanted people to know that there is a Rust language wrapper for libbiio. It doesn't have 100% coverage yet, but with this morning's v0.2 release, it's getting close. Sources are up on GitHub, and the package is published to crates.io.

https://github.com/fpagliughi/rust-industrial-io
https://crates.io/crates/industrial-io/0.2.0

rgetz, 2020-01-03:
Cool - is your plan to send a pull request? or just maintain this separately?

fpagliughi, 2020-01-04:
Oh, I hadn't thought of that. I'm open to suggestions. Would you folks want it?

I had been thinking to see if the Rust Embedded Workgroup would accept it. Either way, though, I'm about to start using it in a commercial product, so am on the hook to make sure it's production ready and maintained for the next few years.

rgetz, 2020-01-07:
We have alot of bindings in the repo already - so as long as it fits into that framework - I don't think it should be a problem. It's more up to @pcercuei - (Paul - any comments?)

pcercuei, 2020-01-07:
I have a small preference with having the Rust bindings in the main repository as well; I think it would have a bit more visibility here.

fpagliughi, 2020-01-07:
I'm cool with that. It's a pretty straight-up wrapper for the library so it should "fit".

I probably have a few more weeks of tweaking it before a good initial release is ready. After that, the only future enhancement I haven't done yet is to have the Rust builder (cargo) automatically compile and link in the C library. Basically, it can automatically run CMake as part of the Rust build.

Then I would need to figure out how to transfer or share ownership of the project on the Rust package site, [crates.io](https://crates.io/crates/industrial-io/0.2.0).  That's how most Rust users would likely download it.

rgetz, 2020-01-10:
It looks like we can use the existing build infrastructure (that we already know) 

Build:
https://docs.travis-ci.com/user/languages/rust/ 

Deploy:
https://docs.travis-ci.com/user/deployment/cargo/

?

fpagliughi, 2020-01-10:
Yes. Just to clarify... 

Cargo is the standard way to build Rust code. If someone want to use IIO in their Rust application, that can just add a single line to their app's cargo file, like:
```
industrial-io = "0.2"
```
This will pull the requested version of the Rust library down from crates.io, build it, and link it into the user's application. (This is amazing to me, coming from a C++ background). 

But this just builds the Rust wrapper library. As it stands now, the user still needs to get the C library sources, compile them manually, and install them on the development host; and then cross-compile them for the target, etc.

The neat thing is that you _could_ also have Cargo grab the C sources from git, compile them for the target (cross-compiling when necessary) by running CMake, generate the proper Rust bindings for the target, and then link the C lib into the Rust app. Completely hands-off.

Meaning... the Rust user just builds their app, and the process downloads and builds both the IIO Rust wrapper _and_ the C lib.

You write the builder to do all this in Rust. No separate Make/build language to learn. This is an example from my Rust MQTT library (which wraps a C MQTT lib):
https://github.com/eclipse/paho.mqtt.rust/blob/master/paho-mqtt-sys/build.rs

rgetz, 2020-01-11:
Yeah, I get that - the bindings and the library are separate (similar for python; where the bindings are installed via pip; and the library (the so) is installed via apt-get; as long as the bindings error nicely when the so in not installed - is should be fine.)

rgetz, 2020-08-27:
Now that this is up in the doc - I will close this issue. If you need anything in this repo - please feel free to make a PR.

closed 2020-08-27 by rgetz