analogdevicesinc/libiio #330: Libiio problem with Libxml2 using prefix

paholden, 2020-01-02:
Was there any resolution to this problem? I am running into a cmake failure when trying to cross-compile libraries.

make
Scanning dependencies of target iio
[ 2%] Building C object CMakeFiles/iio.dir/backend.c.o
[ 5%] Building C object CMakeFiles/iio.dir/channel.c.o
[ 8%] Building C object CMakeFiles/iio.dir/device.c.o
[ 11%] Building C object CMakeFiles/iio.dir/context.c.o
[ 13%] Building C object CMakeFiles/iio.dir/buffer.c.o
[ 16%] Building C object CMakeFiles/iio.dir/utilities.c.o
[ 19%] Building C object CMakeFiles/iio.dir/scan.c.o
[ 22%] Building C object CMakeFiles/iio.dir/sort.c.o
[ 25%] Building C object CMakeFiles/iio.dir/local.c.o
[ 27%] Building C object CMakeFiles/iio.dir/network.c.o
[ 30%] Building C object CMakeFiles/iio.dir/xml.c.o
In file included from /usr/include/libxml2/libxml/parser.h:810:0,
from /usr/include/libxml2/libxml/globals.h:18,
from /usr/include/libxml2/libxml/threads.h:35,
from /usr/include/libxml2/libxml/xmlmemory.h:218,
from /usr/include/libxml2/libxml/tree.h:1307,
from /home/patricaholden/williamson/testbuild/Pro3-Linux/packages/libiio/xml.c:23:
/usr/include/libxml2/libxml/encoding.h:31:26: fatal error: unicode/ucnv.h: No such file or directory
#include <unicode/ucnv.h>
^
compilation terminated.
CMakeFiles/iio.dir/build.make:302: recipe for target 'CMakeFiles/iio.dir/xml.c.o' failed
make[2]: *** [CMakeFiles/iio.dir/xml.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/iio.dir/all' failed
make[1]: *** [CMakeFiles/iio.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

rgetz, 2020-01-07:
It looks like it is looking in host xml2, not your cross compile xml2 (assuming you have that installed?)

Are you building by hand, or building in a build environment (buildroot, yocto, OpenWRT/LEDE, etc?)

pcercuei, 2020-01-07:
How do you call CMake?

paholden, 2020-01-07:
I have the cross compiler installed and configured in the CMakeLists.txt, I'm attaching it here.   I am building by hand because I already have a pre-built Ubuntu image that TI supplies for their Linux SDK.   I just installed crosstool-ng to see if I can use it.  I'm not a big CMake user, predominantly use autotools, so perhaps I messed something up when I added cross-compile hooks.

[CMakeLists.txt](https://github.com/analogdevicesinc/libiio/files/4031914/CMakeLists.txt)

paholden, 2020-01-07:
I made a couple minor changes to my cross-compile version of the CMakeLists.txt file.  I've attached the file and appended the output here.  It seems that pthreads is not found.  I have the sysroot set correctly, but it seems like it is still trying to look for pthreads in /usr/...  Also, it looks like the CC & C++ compiler is still using /usr/bin instead of the cross-compiler path that is defined.

[CMakeLists.txt](https://github.com/analogdevicesinc/libiio/files/4032360/CMakeLists.txt)

cmake command sequence:
=====================
cd libiio
mkdir build
cd build
cmake --sysroot=${PWD}/../../../ubuntu-18.04.2-minimal-armhf-2019-02-16 ../

Output from cmake:
===============
-- The C compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test HAS_WPEDANTIC
-- Performing Test HAS_WPEDANTIC - Failed
-- Looking for strdup
-- Looking for strdup - not found
-- Looking for strerror_r
-- Looking for strerror_r - not found
-- Looking for newlocale
-- Looking for newlocale - not found
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- Looking for libusb-1.0 : Found
-- Looking for libusb_get_version
-- Looking for libusb_get_version - not found
-- Performing Test HAS_O_TMPFILE
-- Performing Test HAS_O_TMPFILE - Failed
-- Performing Test WITH_NETWORK_EVENTFD
-- Performing Test WITH_NETWORK_EVENTFD - Failed
-- Performing Test HAS_PIPE2
-- Performing Test HAS_PIPE2 - Failed
-- Found LibXml2: /home/patricaholden/williamson/testbuild/Pro3-Linux/packages/libiio/../../ubuntu-18.04.2-minimal-armhf-2019-02-16/lib/libxml2.so (found version "2.9.4") 
-- LIBXML2 found with version string
-- Looking for pthread.h
-- Looking for pthread.h - not found
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Threads (missing: Threads_FOUND)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.10/Modules/FindThreads.cmake:205 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:254 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/patricaholden/williamson/testbuild/Pro3-Linux/packages/libiio/build/CMakeFiles/CMakeOutput.log".
See also "/home/patricaholden/williamson/testbuild/Pro3-Linux/packages/libiio/build/CMakeFiles/CMakeError.log".
make: *** No rule to make target 'all'.  Stop.
make: *** No rule to make target 'install'.  Stop.

pcercuei, 2020-01-07:
The proper way to cross-compile with CMake, is to have a toolchain file:

Example toolchain.cmake:
```
set(CMAKE_SYSTEM_NAME Linux)
set(CMAKE_C_COMPILER /opt/foo-toolchain/usr/bin/mipsel-foo-linux-uclibc-gcc)
set(CMAKE_CXX_COMPILER /opt/foo-toolchain/usr/bin/mipsel-foo-linux-uclibc-g++)
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -pipe -O2 " CACHE STRING "Buildroot CFLAGS" FORCE)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -pipe -O2 " CACHE STRING "Buildroot CXXFLAGS" FORCE)
set(CMAKE_INSTALL_SO_NO_EXE 0)
set(CMAKE_PROGRAM_PATH "/opt/foo-toolchain/usr/bin")
set(CMAKE_FIND_ROOT_PATH "/opt/foo-toolchain/usr/mipsel-foo-linux-uclibc/sysroot")
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(ENV{PKG_CONFIG_SYSROOT_DIR} "/opt/foo-toolchain/usr/mipsel-foo-linux-uclibc/sysroot")
```

Then you can cross-compile with the unmodified CMakeLists.txt by doing:
```cmake -DCMAKE_TOOLCHAIN_FILE=toolchain.cmake ..```

paholden, 2020-01-07:
Ok, did not know that.  This will help alot, especially where libiio is not the only package I need to install for this target.  Thanks much!

pcercuei, 2020-01-07:
Where is your libpthread.so?

About the libxml issue: your toolchain's libxml seems to be improperly configured; edit ```<libxml/xmlversion.h>``` and comment the ```LIBXML_ICU_ENABLED``` define.

You shouldn't have to do that hack anyway, maybe recompile libxml2?

rgetz, 2020-01-10:
getting the cross working is fine - but If this is for beagle - why not just compile natively on that platform?

rgetz, 2020-01-10:
Also - per the doc at https://cmake.org/cmake/help/latest/manual/cmake-toolchains.7.html#cross-compiling

Do you have both set:
```
set(CMAKE_SYSROOT /home/devel/rasp-pi-rootfs)
set(CMAKE_STAGING_PREFIX /home/devel/stage)
```

paholden, 2020-01-10:
We don't want to compile natively on the platform as this is going to be a build to put onto many customer boxes.  So prepackaging the entire rootfs is what we want to do, using a standard Intel-based PC.
We are using the TI Beaglebone-X15.  It has its own rootfs, kernel, u-boot already set up with tools.  We are appending to the rootfs that is then programmed to the board.
On the libpthread, it is located in "base-rootfs-path/lib/arm-linux-gnueabihf/", where "base-rootfs-path" is set.
I just checked for CMAKE_SYSROOT and CMAKE_STAGING_PREFIX.  They are not set up.  Thanks for the doc, not sure I found that in my many google searches while working on this project.
I did do a work-around for now, which was to just run "dpkg" on the target & try to use the raspbian canned package latest-release.  That runs on the target.  But this is just a bandaid so we can get back to debugging my driver instead of messing with build tools.
Thanks for the help, going to retry after adding the 2 missing definitions.

rgetz, 2020-01-11:
@paholden : understand your issues - while it's nicer to build on desktop - not enough people understand cross build properly to write their upstream cmake (sometimes) - it was always worse with autotools (more people made more mistakes); and setting up everything is a gain pain sometimes.

The X15 root fs is based on Debian 10 (which already has libiio in it).https://tracker.debian.org/pkg/libiio - so standard apt-get should work for almost every arch.

If fixing the CMAKE_SYSROOT doesn't help - let us know.

-Robin

paholden, 2020-01-13:
I will let you know how I solve it, one way or another, just so others who run into problems can have a solution.  Thanks for your continued support!

rgetz, 2020-01-28:
Ok - I will close this for now. If you still run into issues, please let us know, by the-opening this thread.

closed 2020-01-28 by rgetz

paholden, 2020-01-28:
Ok,  thank you for your assistance to date.  I will post the solution as soon as I get back to this one.  For now, while debugging my driver, I am just using "sudo apt install" to get libiio dependencies, adding "master_latest_libiio-raspbian-8-armhf.deb" to the SD rootfs build script, and running "dpkg -i master_latest_libiio-raspbian-8-armhf.deb" once the target is running.